AltBeacon/android-beacon-library #323: Exception that actives with a massive number of Bluetooth signals (beacon and smatphone)

Kain86, 2015-12-13:
Tested with more devices (samsung smartphone and tablet with OS 4.4.2 and 5) when start scan, the app, after few seconds, crashes with this error:

java.lang.RuntimeException: An error occured while executing doInBackground()
	at android.os.AsyncTask$3.done(AsyncTask.java:300)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
	at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
	at java.util.concurrent.FutureTask.run(FutureTask.java:242)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
	at java.lang.Thread.run(Thread.java:818)
Caused by: java.lang.ArrayIndexOutOfBoundsException: length=62; index=62
	at org.altbeacon.beacon.BeaconParser.byteArraysMatch(BeaconParser.java:777)
	at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:416)
	at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:373)
	at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:457)
	at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:442)
	at android.os.AsyncTask$2.call(AsyncTask.java:288)
	at java.util.concurrent.FutureTask.run(FutureTask.java:237)

davidgyoung, 2015-12-13:
The stack trace indicates that the library crashed trying to process malformed Bluetooth LE packet.   There is a flaw in line BeaconParser.java:777 in the byteArraysMatch method that overflows an array in certain cases.

```
Caused by: javalangArrayIndexOutOfBoundsException: length=62; index=62
at orgaltbeaconbeaconBeaconParserbyteArraysMatch(BeaconParserjava:777)
at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:416)
```

davidgyoung, 2015-12-13:
Fixed in #324

PR #324 (2015-12-20): Fix crash on comparison overrun 
When comparing two byte arrays, check first to see if one is shorter than the other.  If so, consider them unequal and short-circuit the comparison rather than crash on an ArrayIndexOutOfBoundsException looping through the bytes.  Fixes #323
closed 2015-12-13 by davidgyoung