AltBeacon/android-beacon-library #326: Android 6 (main thread is busy, if we don't have the location permission)

contrudar, 2015-12-17:
Let's imagine the next situation. What we have:
* Android 6 (Nexus 5)
* Backgroud scan period equals 1100 ms
* We don't have the location permission

As intended, SecurityException occurs every second as you can see in LogCat: 
12-17 22:37:10.384: W/Binder(27926): Caught a RuntimeException from the binder stub implementation.
12-17 22:37:10.384: W/Binder(27926): java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
12-17 22:37:10.384: W/Binder(27926): 	at android.os.Parcel.readException(Parcel.java:1599)
12-17 22:37:10.384: W/Binder(27926): 	at android.os.Parcel.readException(Parcel.java:1552)
12-17 22:37:10.384: W/Binder(27926): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:772)
12-17 22:37:10.384: W/Binder(27926): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:324)
12-17 22:37:10.384: W/Binder(27926): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
12-17 22:37:10.384: W/Binder(27926): 	at android.os.Binder.execTransact(Binder.java:453)

It happens in the main thread, so the UI is freezing for almost 2 seconds after each scanning when we encounter SecurityException. 

Of course, we can not start monitor beacons if we don't have the location permission. But what if the user allows/denies the location permission through standard settings not through our UI dialogs. As I know, android doesn't send any intents for this. In case when the user denies the location permission through settings, we won't get any notifications and will get a periodic freezing of the main thread. In case when the user allows the location permission through settings, we also won't get it and can't start monitor again.

What is the decision of this problem? Or maybe I misunderstand something.

Thanks in advance.

davidgyoung, 2015-12-17:
I think this analysis is correct.  Probably the best solution is to modify the library so it verifies the permission each time before starting scanning.  Something like this (pseudocode):

```
if verifyLocationPermissionGranted() {
  startScan()
}
```

davidgyoung, 2015-12-18:
Fixed in #327

closed 2015-12-18 by davidgyoung