AltBeacon/android-beacon-library #329: Guaranteed ArrayIndexOutOfBoundsException in BeaconParser class.

VGJankovsky, 2015-12-25:
I was trying to implement a "getBeaconAdvertisementData" method for iOS version of BeaconParser from [ios-becon-tools](https://github.com/AltBeacon/ios-beacon-tools), so I was looking at java code and found an issue, specifically here:

    // set identifiers
        for (int identifierNum = 0; identifierNum < this.mIdentifierStartOffsets.size(); identifierNum++) {
            byte[] identifierBytes = beacon.getIdentifier(identifierNum).toByteArrayOfSpecifiedEndianness(!this.mIdentifierLittleEndianFlags.get(identifierNum));

            // If the identifier we are trying to stuff into the space is different than the space available
            // adjust it
            if (identifierBytes.length < getIdentifierByteCount(identifierNum)) {
                if (!mIdentifierVariableLengthFlags.get(identifierNum)) {
                    // Pad it, but only if this is not a variable length identifier
                    if (mIdentifierLittleEndianFlags.get(identifierNum)) {
                        // this is little endian.  Pad at the end of the array
                        identifierBytes = Arrays.copyOf(identifierBytes,getIdentifierByteCount(identifierNum));
                    }
                    else {
                        // this is big endian.  Pad at the beginning of the array
                        byte[] newIdentifierBytes = new byte[getIdentifierByteCount(identifierNum)];
                        System.arraycopy(identifierBytes, 0, newIdentifierBytes, getIdentifierByteCount(identifierNum)-identifierBytes.length, identifierBytes.length);
                        identifierBytes = newIdentifierBytes;
                    }
                }
                LogManager.d(TAG, "Expanded identifier because it is too short.  It is now: "+byteArrayToString(identifierBytes));
            }
            else if (identifierBytes.length > getIdentifierByteCount(identifierNum)) {
                if (mIdentifierLittleEndianFlags.get(identifierNum)) {
                    // Truncate it at the beginning for big endian
                    identifierBytes = Arrays.copyOfRange(identifierBytes, getIdentifierByteCount(identifierNum)-identifierBytes.length, getIdentifierByteCount(identifierNum));
                }
                else {
                    // Truncate it at the end for little endian
                    identifierBytes = Arrays.copyOf(identifierBytes,getIdentifierByteCount(identifierNum));
                }
                LogManager.d(TAG, "Truncated identifier because it is too long.  It is now: "+byteArrayToString(identifierBytes));
            }

First is:

    else if (identifierBytes.length > getIdentifierByteCount(identifierNum)) {
                    if (mIdentifierLittleEndianFlags.get(identifierNum)) {
                        // Truncate it at the beginning for big endian
                        identifierBytes = Arrays.copyOfRange(identifierBytes, getIdentifierByteCount(identifierNum)-identifierBytes.length, getIdentifierByteCount(identifierNum));
                    }

 **getIdentifierByteCount(identifierNum) - identifierBytes.length** will always be a negative integer (since it is calculate it case of **identifierBytes.length > getIdentifierByteCount(identifierNum)**, so we have a negative number as "from" argument which leads to an ArrayIndexOutOfBoundsException, according to documentation.

Second is inconsistency in "if" conditions:

Here we have a comment "// this is little endian.  Pad at the end of the array" for a true little endian flag:

    if (mIdentifierLittleEndianFlags.get(identifierNum)) {
                            // this is little endian.  Pad at the end of the array
                            identifierBytes = Arrays.copyOf(identifierBytes,getIdentifierByteCount(identifierNum));
                        }
                        else {
                            // this is big endian.  Pad at the beginning of the array
                            byte[] newIdentifierBytes = new byte[getIdentifierByteCount(identifierNum)];
                            System.arraycopy(identifierBytes, 0, newIdentifierBytes, getIdentifierByteCount(identifierNum)-identifierBytes.length, identifierBytes.length);
                            identifierBytes = newIdentifierBytes;
                        }

And in the code below it it's reversed (for true little endian flag we have a "// Truncate it at the beginning for big endian" comment) :

       if (mIdentifierLittleEndianFlags.get(identifierNum)) {
                        // Truncate it at the beginning for big endian
                        identifierBytes = Arrays.copyOfRange(identifierBytes, getIdentifierByteCount(identifierNum)-identifierBytes.length, getIdentifierByteCount(identifierNum));
                    }
                    else {
                        // Truncate it at the end for little endian
                        identifierBytes = Arrays.copyOf(identifierBytes,getIdentifierByteCount(identifierNum));
                   }

Can you state if those two are bugs, or it was implemented that way for some particular reason.

Thanks in advance.

davidgyoung, 2015-12-25:
Good catches, @VGJankovsky.   These look do look like bugs.  I think new test cases in BeaconParserTest for identifier bytes overruning the length in the layout would help chatch this.  If you are up for it, I would welcome a pull request to fix these issues.

Wihlly, 2016-07-25:
Hi,

I have a similar issue that seems to be related to this post.

```
Fatal Exception: java.lang.RuntimeException
An error occurred while executing doInBackground()
        android.os.AsyncTask$3.done (AsyncTask.java:309)
        java.lang.Thread.run (Thread.java:818) 

Caused by java.lang.ArrayIndexOutOfBoundsException
org.altbeacon.beacon.BeaconParser.byteArrayToFormattedString (BeaconParser.java:796)
org.altbeacon.beacon.BeaconParser.fromScanData (BeaconParser.java:542)
org.altbeacon.beacon.BeaconParser.fromScanData (BeaconParser.java:373)
org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground (BeaconService.java:457)
org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground (BeaconService.java:442)
android.os.AsyncTask$2.call (AsyncTask.java:295)
java.lang.Thread.run (Thread.java:818)
```

Weirdly, this error only occured with Infinix devices on Android M.
I will really appreciate any information for the resolution of this issue.

Thanks

davidgyoung, 2016-07-25:
@WilliamMouline, I suspect this is a different issue.  If you don't mind, can you please add `beaconManager.setDebug(true)` to your code, reproduce the bug, then look for lines like below right before the crash.  You can then paste this into a new issue in this repository and link to it from here.  I'd be glad to investigate if you can collect this info.

```
07-25 14:11:21.710 18995-19819/com.test.test D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
07-25 14:11:21.710 18995-19818/com.test.test D/BeaconParser: Bytes are: 0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa600010001b4000000000000000000000000000000000000000000000000000000000000000
```