AltBeacon/android-beacon-library #331: Support for API 15: Android 4.0.3 (IceCreamSandwich)?

Ghenty, 2016-01-03:
Hi,

I bought a Vuzix SmartGlass and want to use the android beacon library on it - unfortunately this device runs with Android 4.0.3 and I immediately get the error that the BluetoothManager has all methods.

Is there an easy possibilty (some older release i.e.) to this library run on this SmartGlass?

Best regards,
Christian

davidgyoung, 2016-01-04:
@Ghenty, unfortunately, Android introduced Bluetooth LE APIs only in their 4.3 release (SDK 18), so earlier versions of Android do not have the Bluetooth LE APIs needed for this library to work.  I note that Vuzix claims their hardware has Bluetooth 4.0 support (4.0 added Bluetooth LE), so if the hardware actually has the capability to detect Bluetooth LE beacons, you then need an OS image with driver and OS-level support.    Two options here:

1. Try to get a 4.3+ OS image for the device (made by Vuzix or a third party)

2. See if there is a proprietary Bluetooth LE SDK from Vuzix.  (I looked briefly but did not see one.)  Samsung, for example, built a proprietary SDK for their devices running 4.2, and a fork of this library exists that works with it.  If you can find a proprietary SDK from Vuzix, you may be able to get this library to work.

closed 2016-01-07 by davidgyoung

katoqiioo, 2023-09-30:
-- @katoqiioo --

##    Proyecto Webapp   ##

# 1. Android Studio | New Project | Minimum SDK.
2. Selecciona "API 15: Android 4.0.3" o una versión superior y crea tu nuevo proyecto.
3. En el proyecto, abre

your_app | Gradle Scripts | build.gradle
.

4. Agrega el repositorio central de Maven a build.gradle antes de dependencies:.
repositories {
                           mavenCentral()