AltBeacon/android-beacon-library #332: Jake Wharton's SDK Manager plugin does not work with Android Gradle plugin 2.0.0-alpha1

ma-rp, 2016-01-08:
When updating to build-plugin to 2.0.0-alpha1 the following error comes up:
Error:Cannot cast object '23.0.2' with class 'com.android.repository.Revision' to class 'com.android.sdklib.repository.FullRevision'﻿

You can see its related to Jake Wharton's SDK Manager plugin, here are the details:

http://stackoverflow.com/questions/33881984/errorcause-com-android-sdklib-repository-fullrevision/33889117#33889117

https://github.com/JakeWharton/sdk-manager-plugin/pull/100

RBWare, 2016-05-26:
This is an issue with Jake Wharton's library, not with AltBeacon's. There are several open issues on his in the SDK Manager repo. Even the bug you referenced points to his repo as the issue. It has been fixed, but he has not (and according to him, will not) publish and update to that library. The SDK Manager library was never completed and fully working anyway. The recommendation is to remove that from your project.

davidgyoung, 2016-05-31:
Thanks, @RBWare.  I agree the best course of action is to remove this so we are compatible with newer gradle versions.

davidgyoung, 2016-06-24:
Plugin removed in #394

closed 2016-06-24 by davidgyoung