AltBeacon/android-beacon-library #335: Using RTC timer breaks periodic scanning when clock is adjusted

lultimouomo, 2016-01-29:
I have been experiencing a bug on a device without battery backed RTC:
if the device is started without a network connection, and then the network cable is attached later on, beacon ranging stops - i.e., didRangeBeaconsInRegion is never called again.

I think this is due to the fact that at startup the RTC is set in the future (yes it is...), and when the net is connected NTP makes it jump back to the present.
CycledLeScanner uses AlarmManager.RTC_WAKEUP, which probably breaks when the RTC clock is adjusted. Is there a reason for not using AlarmManager.ELAPSED_REALTIME_WAKEUP?

davidgyoung, 2016-01-30:
The library actually use to to use ELAPSED_REALTIME_WAKEUP, but it was changed to solve a different problem mentioned in #154.  

While the problem you mention is not common, it is possible that it could be easily fixed with a change from:

```
alarmManager.set(AlarmManager.RTC_WAKEUP, System.currentTimeMillis() + milliseconds, getWakeUpOperation());
```

to:

```
alarmManager.set(AlarmManager.ELAPSED_REALTIME_WAKEUP, System.elapsedRealtime() + milliseconds, getWakeUpOperation());
```

Would you be interested in making an testing this change?  If it works, a pull request would be welcome.

lultimouomo, 2016-01-30:
I can give it a try next week.

davidgyoung, 2016-02-15:
Closed via #339

closed 2016-02-15 by davidgyoung