AltBeacon/android-beacon-library #337: Update README: Android M Runtime Permissions

cavega, 2016-02-01:
The README should be updated to indicate that developers targeting Marshmallow (SDK 23) must request the following runtime permission before been able to use the library for scanning beacon devices:

```
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
```

Otherwise, the library won't work and the app will registered the following in logcat:

```
W/Binder: Caught a RuntimeException from the binder stub implementation.
    java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
    at android.os.Parcel.readException(Parcel.java:1620)
    at android.os.Parcel.readException(Parcel.java:1573)
    at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:772)
    at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:324)
    at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
    at android.os.Binder.execTransact(Binder.java:453)
```

crespum, 2016-04-05:
Actually the solution is not what you propose. You have to ask the user for permission to access the location info. You can check all the details [in the project website](https://altbeacon.github.io/android-beacon-library/requesting_permission.html).

aldoborrero, 2016-04-14:
To be more precise the library does include the required permission by default if the api is 23 or superior:

https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/AndroidManifest.xml#L8

See that.

As @crespum has wrote you only have to prompt the user to use enable it

davidgyoung, 2016-04-14:
This all sounds right.  I think this issue really is that it is easy for folks to forget to do that.  I'm not sure if a reminder in the README would help with that, but it certainly wouldn't hurt.   I am open to other suggestions on how to make this easier on folks first working with the library.

cavega, 2016-04-14:
Yes, the library has the permission included in the manifest.  My original comment was more along the lines of "hey developer, remember to show the request permission prompt to your users". From what I see now in the README there is a comment suggesting that.

closed 2016-07-22 by davidgyoung