AltBeacon/android-beacon-library #338: Cannot make foreground status and background monitor work together

melson-jao, 2016-02-02:
Hi,

I am not sure it is an issue or not. Anyway, my problem is:
I implement an activity followed by example and it is okay to detect beacon and trigger didEnterRegion successfully.
public class MonitoringActivity extends Activity implements BeaconConsumer {

And then I implement a background application  and it is okay to detect beacon and trigger didEnterRegion in background.
public class BackgroundActivity extends Application implements BootstrapNotifier {

But when I try to combine them together, they cannot work.
What I want to do is:
When user launch the app, he can see the detected beacon in region and when he pause the app, the background application will keep monitoring beacon. In case a new beacon appears, the background app will detect it and push local notification to user.

My test tells me:
1. when I have "regionBootstrap = new RegionBootstrap(this, regionList);" in BackgroundActivity onCreate(),  the MonitoringActivity shows nothing even if beacon is advertising. After I remove regionBootStrap register, MonitoringActivity shows beacon correctly.
2. In previous case, another way to make MonitoringActivity show beacon correctly is after I pause the app (go to background) and resume it (back to foreground) again and then MonitoringActivity didEnterRegion triggered correctly.
3. I try to push back regionBootstrap register from app launch time to later, ex MonitoringActivity onPause(). In this case, after app launch, MonitoringActivity show beacons correctly. And when I pause it to background, the startBeaconMonitoring() called which do the regionBootstrap register. but in this case, the background task cannot detect the beacon didEnterRegion.

I guess it is caused by one beaconManager in BackgroundActivity and another beaconManager in MonitoringActivity but I am not sure.

Any comment is appreciated about how to do what I want to do.

Here is my AndroidManifest.xml

<application
        android:name="com.abc.BackgroundActivity"
        android:allowBackup="true"
        android:icon="@mipmap/ic_launcher"
        android:label="@string/app_name"
        android:theme="@style/AppTheme" >
        
        <activity
            android:launchMode="singleInstance"
            android:name="com.abc.MonitoringActivity"
            android:label="@string/app_name" >
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />
                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
        </activity>
    </application>

davidgyoung, 2016-02-02:
@melson-jao, happy to help with this, but can you please create a question on stackoverflow.com?  This forum is reserved for feature requests and bug reports with the library itself.

In general, you should only use RegionBootstrap with a central Application class, not an Activity.

closed 2016-02-02 by davidgyoung

melson-jao, 2016-02-04:
Sure. Got it. Sorry for any inconvenience.