AltBeacon/android-beacon-library #341: Why is scanning needed?

andersborgabiro, 2016-02-15:
This is not an issue per se (battery-drain-wise it could be though), rather a question about why it works the way it does. Remove if deemed off-topic.

I can't quite figure out why BLE scanning needs to be done when listening for beacons. A beacon packet is sent as a broadcast from the beacon without the receiving device transmitting anything back and without pairing being done. Why is it then that scanning needs to be made at all? Is that how Bluetooth fundamentally works to at all hear/acknowledge anything from a (possibly unknown) device?

Cheers,
Anders

davidgyoung, 2016-02-15:
@andersborgabiro, yes, as you say, this is "how Bluetooth fundamentally works."  In order for a mobile device to look for BLE beacons, it must have its bluetooth radio receiver turned on and listening on multiple channels where BLE advertisements may be found.  Putting a bluetooth chip into a mode that does this is essentially what we man by performing a BLE scan.    

Surprising as it may seem, simply having the radio on and the chip actively decoding everything it hears to see if they are Bluetooth LE advertisements actually uses more battery than transmitting a beacon advertisement packet regularly.  I'm not enough of a bluetooth expert to explain *why* this is true, I just know from experience that it is.  You might look to get an answer to the *why* part of this question in a dedicated forum like http://electronics.stackexchange.com/

closed 2016-02-15 by davidgyoung

andersborgabiro, 2016-02-15:
I guess continuous listening vs occasional transmission is one reason. Maybe also to achieve the very high sensitivity of BLE.

It clarifies a bit, as scanning otherwise could mean an active part on the transmitter, asking for devices to respond. That should not be needed here in any case.

Thanks,
Anders

andersborgabiro, 2016-02-15:
I found this, that probably covers most of the possible questions, even those I don't know yet:

http://www.aislelabs.com/reports/ibeacon-battery-phones/