AltBeacon/android-beacon-library #343: No beacon detection if the Android device OS is lower than 5

tyler77, 2016-02-23:
Here is the code that we're using:

`

public class ApplicationClass extends Application implements BootstrapNotifier, BeaconConsumer {
	private static final String TAG = "Demo1";
	private RegionBootstrap regionBootstrap;
	private BeaconManager beaconManager;
	@Override
	public void onCreate()
	{
		super.onCreate();

		// Initialize the singletons so their instances
		// are bound to the application process.
		initSingletons();
	}

	public String getIncompatibilityReason(Context context) {
		if (android.os.Build.VERSION.SDK_INT < 18) {
			return "requires Android 4.3";
		}
		if (!context.getPackageManager().hasSystemFeature(PackageManager.FEATURE_BLUETOOTH_LE)) {
			return "requires Bluetooth LE";
		}
		return null;
	}  

	protected void initSingletons()
	{
		
		BeaconManager.setsManifestCheckingDisabled(true);
		Log.d(TAG, "App started up");
		beaconManager = BeaconManager.getInstanceForApplication(this);
		beaconManager.bind(this);
		// wake up the app when any beacon is seen (you can specify specific id filers in the parameters below)
	
	}
	
	@Override
	public void onBeaconServiceConnect() {
		// TODO Auto-generated method stub
		beaconManager.setBackgroundBetweenScanPeriod(10000l);
	    beaconManager.setBackgroundScanPeriod(10000l);
	    beaconManager.setBackgroundMode(true);
	      try {
			beaconManager.updateScanPeriods();
		} catch (RemoteException e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		}
	       beaconManager.getBeaconParsers().add(new BeaconParser().
	              setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));
	   	Region region = new Region("com.easibeacon.demos.demo1", null, null, null);
		regionBootstrap = new RegionBootstrap(this, region);
		

	}

	

	public void showNotification(String title, String message) {
		Intent notifyIntent = new Intent(this, BeaconListActivity.class);
		notifyIntent.setFlags(Intent.FLAG_ACTIVITY_SINGLE_TOP);
		PendingIntent pendingIntent = PendingIntent.getActivities(this, 0,
				new Intent[] { notifyIntent }, PendingIntent.FLAG_UPDATE_CURRENT);
		Notification notification = new Notification.Builder(this)
		.setSmallIcon(android.R.drawable.ic_dialog_info)
		.setContentTitle(title)
		.setContentText(message)
		.setAutoCancel(true)
		.setContentIntent(pendingIntent)
		.build();
		notification.defaults |= Notification.DEFAULT_SOUND;
		NotificationManager notificationManager =
				(NotificationManager) getSystemService(Context.NOTIFICATION_SERVICE);
		notificationManager.notify(1, notification);
	}

	@Override
	public void didDetermineStateForRegion(int arg0, Region arg1) {
		// TODO Auto-generated method stub

	}

	@Override
	public void didEnterRegion(Region arg0) {
	
		showNotification(
				"Enter in Beacon area",
				"Thanks");
		Intent intent = new Intent(this, BeaconListActivity.class);

		intent.setFlags(Intent.FLAG_ACTIVITY_NEW_TASK);
		this.startActivity(intent);

	}

	@Override
	public void didExitRegion(Region arg0) {
		showNotification(
				"Exit From Beacon",
				"Thanks");
	}

}

`
We have made some tests on the following devices:
Samsung Neo Plus - Android 4.4.4
Sony Xperia Z1 - Android 4.3
Huawei Y125U21 - Android 4.4.2
Huawei P8 - Android 5

Only the P8 detects the beacons and detects the region also in background.
On stackoverflow I found this post and it seems my problem..
http://stackoverflow.com/questions/35503904/android-4-4-ble-scanning-lack-of-stability

What do you think? Do you think it is an OS problem? Any advice?
Please, can you check if my code is correct?
Thank you very very much.

tyler77, 2016-02-23:
I've made other tests with 2.6.1, and always seems to work properly. Tomorrow, I'll do other test and update this post.

davidgyoung, 2016-02-24:
Are you saying it works properly with 2.6.1 but not a newer version?

davidgyoung, 2016-03-14:
@tyler77, please advise if you are still seeing an issue.  It is unclear to me which of the following is the problem:

1. you are seeing foreground detections on 4.x but not background detections.
2. You are seeing no detections on 4.x in either the foreground or the background
3. You are seeing detections on 4.x in with 2.6.1, but not with the latest Library version

tyler77, 2016-04-01:
Hi David, sorry for my radio silence, but i wanted to analyzed the situation before writing something. First of all, here we're working on Eclipse, and the above code is still the one we're testing. 
From the last time we have written, we have never been able to start background monitoring, so I don't know if the problem is related to the 6.2.1. version or not, becouse even with 2.6.1. the code does not work for us.
We have followed the instruction to configure the project (library import - manifestmerger)  and my manifest on bin folder says that merge was successfully done. Foreground detection works like a charm.
I don't know what to check to fix the issue.
I've searched on internet everything related to find out what I'm missing but really can't understand.
Have you ever experimented a case when the foreground monitoring works and background not?
I've uploaded a sample Eclipse Project to test.
Thanks. 
[testbeacon.zip](https://github.com/AltBeacon/android-beacon-library/files/199671/testbeacon.zip)

Should I open a post on stackoverflow?

davidgyoung, 2016-04-01:
Thanks for the clarification.  Yes, it seems this is an Eclipse-specific project configuration issue.  I'd be happy to help on StackoveFlow, but before you open a new question, you might want to check out this one here:  http://stackoverflow.com/questions/26799068/i-cant-detect-ibeacon-in-background-using-altbeacon-library-in-android-app

If that doesn't help, please post a new question on StackOverflow.com and show the setup of your AndroidManifest.xml

closed 2016-04-01 by davidgyoung

tyler77, 2016-04-04:
I read the post but unfortunately it doesn't help.
I launch the project also on Android Studio.
Here is the question on stackoverflow:

http://stackoverflow.com/questions/36399518/altbeacon-doesnt-detect-beacon-when-app-is-in-background

really hope to find a solution..

Best