AltBeacon/android-beacon-library #344: Library Suddenly Stop Working

joginder89, 2016-02-26:
Application is working fine suddenly it stop working and giving  
https://gist.github.com/joginder89/1fda79dfe6ba5afce2cb

davidgyoung, 2016-02-26:
I don't see anything in the gist, just:

```
gistfile1.txt
Can't reach server
```

joginder89, 2016-02-26:
I have re-format the gist. The text was on right hand side.
https://gist.github.com/joginder89/1fda79dfe6ba5afce2cb

because previously it was working fine, it may be issue of internet but not sure. Whether it require to connect to data.altbeacon.org:80  hostAddress=184.73.148.59 to start working.

davidgyoung, 2016-02-26:
That exception is caused by an error in the network connection used to fetch an updated database of Android device-specific parameters for distance estimates.  If a call to fetch the database update fails, the library is designed to simply continue working and use the built-in database.   In this case, the exception shown in the logs should not mean the library or app crashes -- it just gives an indication that the database update failed. 

Do you have any reason to believe that the exception is causing the library to stop detecting beacons?

joginder89, 2016-02-27:
I install the application on a device and test, library give this error and not detect the beacon. Then I test on a device on which my application was install, that was working fine. Then after some time I uninstall the application on which library was working and again install the application. Then that device also start giving that exception and stop detecting beacon. Even my previous apps also stop detecting beacon.

I think when we install the application, then one time library require to connect to data.altbeacon.org:80 to fetch the database or database is already present inside the library.

joginder89, 2016-02-27:
Today again I check, its start working again on Nexus 7 but not works on other mobiles. It gives 
    W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
then I restart phone and re-connect charger after that its start working again. Meanwhile I visit data.altbeacon.org from my browser It giving very slow response, take almost 3 minutes to load web page while other sites are working fine (internet connection was not slow). So it was only network connection problem.

Thank you for your time.

closed 2016-02-27 by joginder89