AltBeacon/android-beacon-library #345: Implementing BeaconConsumer as a Service

lkwjohn, 2016-03-01:
I have been trying to implement the BeaconConsumer as a Service but 
when it comes to the beaconManager.bind(this) I always get 
java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String android.content.Context.getPackageName()' on a null object reference

org.altbeacon.beacon.BeaconManager.bind(BeaconManager.java:306)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at BeaconServiceTest.setupBeaconManager(BeaconServiceTest.java:279)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at BeaconServiceTest.onCreate(BeaconServiceTest.java:267)

From what it seems, there was a null at BeaconManager line 306.

The only line work was the beaconManager = BeaconManager.getInstanceForApplication(this);

the AltBeacon library version is 2.7.1

lkwjohn, 2016-03-01:
This is the full trace

03-01 18:10:10.241: E/AndroidRuntime(4862): java.lang.RuntimeException: Unable to create service sg.edu.smu.livelabs.conferenceapp.BeaconServiceTest: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String android.content.Context.getPackageName()' on a null object reference
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at android.app.ActivityThread.handleCreateService(ActivityThread.java:3290)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at android.app.ActivityThread.access$1900(ActivityThread.java:181)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1565)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at android.os.Handler.dispatchMessage(Handler.java:102)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at android.os.Looper.loop(Looper.java:145)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at android.app.ActivityThread.main(ActivityThread.java:6145)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at java.lang.reflect.Method.invoke(Native Method)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at java.lang.reflect.Method.invoke(Method.java:372)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)
03-01 18:10:10.241: E/AndroidRuntime(4862): Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String android.content.Context.getPackageName()' on a null object reference
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at android.content.ComponentName.<init>(ComponentName.java:77)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at android.content.Intent.<init>(Intent.java:4618)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at org.altbeacon.beacon.BeaconManager.bind(BeaconManager.java:306)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at sg.edu.smu.livelabs.conferenceapp.BeaconServiceTest.setupBeaconManager(BeaconServiceTest.java:279)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at sg.edu.smu.livelabs.conferenceapp.BeaconServiceTest.onCreate(BeaconServiceTest.java:267)
03-01 18:10:10.241: E/AndroidRuntime(4862): 	at android.app.ActivityThread.handleCreateService(ActivityThread.java:3280)

davidgyoung, 2016-03-01:
I suspect this has to do with the context being null in the `BeaconConsumer` implementation.  Please post this as a question on StackOverflow as that is the preferred forum for project-specific support questions.  This forum is reserved for feature requests and reproducible bugs with the library itself.

In your question, please include a code snippet showing the `getApplicationContext()` implementation of your `BeaconConsumer` implementation.

closed 2016-03-01 by davidgyoung