AltBeacon/android-beacon-library #348: How to add extra string in AltBeacon's layout for advertising?

mbmc, 2016-03-10:
I'm trying to add an extra string in a beacon, but not sure how to proceed. The data field only takes a `List` of `Long`. Should I modify the layout's format? What if the string's length is variable?

davidgyoung, 2016-03-12:
The beacon layout's data field prefix is not designed to contain data types like strings with arbitrary byte lengths.  In its current implementation it is typically used for single byte fields.  Currently the library's beacon layout format supports arbitrary length fields only for identifier fields (the "i" prefix).  The layout expression "i:1-10v" defines an identifier field that is between 1 and 10 bytes long.  If endcoding string data inside an identifier field, you will need to write your own enoding/decoding function to convert it to bytes.  This is what is done for Eddystone-URL beacons, which essentially have a string encoded in their  single identifier field with a custom compression/decompression algorithm implemented here:  https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/utils/UrlBeaconUrlCompressor.java

closed 2016-03-12 by davidgyoung

andersborgabiro, 2016-03-23:
I'm wondering what the string would do. What do you expect would happen? No existing app would be compatible with that custom format.

The most straightforward option is to write a URL to an Eddystone beacon and then put the ad and other content on the accessed site instead.

Yet, we offer a solution that kind of works like Eddystone-URL, but using iBeacon via a Physical Web-like app, and in a much more flexible way (many beacons per site, many sites per beacon etc).

zhubinsheng, 2021-11-17:
> 直接的选择是将 URL 写入 Eddystone 信标，然后将广告和其他内容放在访问的站点

hello Is there a help website with more information about what you said