AltBeacon/android-beacon-library #349: After a while, scan stops and Bluetooth freeze. Need reboot smartphone

jvidalgz, 2016-03-12:
Hello, I am working with a simple code, that scan iBeacons, then I store them in a ArrayList, but after a cople minutes the scan stop and I can use  the bluetooth, even anothers ibeacon scanner applications dont work. If I reboot smarthphone this is fixed.

`public class MainActivity extends Activity implements BeaconConsumer {
    protected static final String TAG = "MonitoringActivity";
    private BeaconManager beaconManager;
    ArrayList<Beacon> mBeaconsStorage = new ArrayList<>();

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        //setContentView(R.layout.activity_ranging);
        beaconManager = BeaconManager.getInstanceForApplication(this);

        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.setDebug(true);
        beaconManager.bind(this);
    }
    
    @Override
    protected void onDestroy() {
        super.onDestroy();
        mBeaconsStorage.clear();
        beaconManager.unbind(this);
    }

    @Override
    public void onBeaconServiceConnect() {
        // set the duration of the scan to be 10.1 seconds
        beaconManager.setBackgroundScanPeriod(11000l);
        // set the time between each scan to be 1.1 seconds
        beaconManager.setBackgroundBetweenScanPeriod(1100l);

        beaconManager.setRangeNotifier(new RangeNotifier() {

            @Override
            public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
                //Log.d(TAG, "Total beacons: " + String.valueOf(beacons.size()));
                if (beacons.size() > 0) {
                    for (Beacon b : beacons) {
                        processYourBeaconInThisMethod(b);
                    }
                    Log.d(TAG, "Total ibeacons en mIbeaconStorage: " + String.valueOf(mBeaconsStorage.size()) + " ibeacons.");
                }
            }
        });      

        try {
            beaconManager.startRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
        } catch (RemoteException e) {
        }

    }  

    public void processYourBeaconInThisMethod(Beacon beacon) {
        mBeaconsStorage.add(beacon);
    }

}
`

http://pastebin.com/gFhwinYY, in the first line I got the last ibeacon, then nothing.

davidgyoung, 2016-03-12:
This looks like a duplicate of #289.  Can you please tell me your phone model and exact OS version and build string to confirm?  Also, the pastebin link in the issue is private, so can you please update it to be public?

jvidalgz, 2016-03-12:
OK.

I'm using a Huawei Y625 smartphone with Android 4.4.4.
Pastebin link is now public.
By build string , you mean 'kernel version' or not? I'm a lost

davidgyoung, 2016-03-12:
You can get the build number string from Settings -> Abut phone -> Build number.  

It would be helpful to see a log except that:

* spans several scan cycles, at least 60 seconds long
* contains system level log messages (is not filtered for your app).  This would allow seeing any system-level bluetooth messages.

jvidalgz, 2016-03-12:
Here is my new pastebin file (with no filters)

http://pastebin.com/PzUwv4Yk

Just for testing purposes, I was counting the numbers of advertising packes stored in the ArrayList. I hit 787 'ibeacons' (from the same fisical device - (adafruit BLE shield)) but then nothing, the scan stop.

In line 41 of the pastebin file I got 786 ibeacons and the last one in the line 80. The line 356 show  ' This is not Android 5.0.  We are using old scanning APIs', is that relevant or not?

Here is a screenshot from About phone section: http://postimg.org/image/4zvxbm6oz/

I found this commit from android.googlesource.com: https://android.googlesource.com/platform/system/bt/+/ed1563b
and I think is relate to this issue.

jvidalgz, 2016-03-12:
Here is my new pastebin file (with no filters)

http://pastebin.com/JK5czCd7

Just for testing purposes, I was counting the numbers of advertising packes stored in the ArrayList. I hit 652 'ibeacons' (from the same fisical device - (adafruit BLE shield)) but then nothing, the scan stop.

In line 1 of the pastebin file I got 651 ibeacons and the last one in the line 81. The line 1533 show 'This is not Android 5.0. We are using old scanning APIs', is that relevant or not?

Here is a screenshot from About phone section: http://postimg.org/image/4zvxbm6oz/

I found this commit from android.googlesource.com: https://android.googlesource.com/platform/system/bt/+/ed1563b
and I think is relate to this issue.

davidgyoung, 2016-03-12:
In the most recent referenced log. I see an app crashed at the referenced lines below.  Is this your app or a different app?  It appears no BLE detections happen after this.

``03-12 11:26:53.905 3317-3323/cl.qualitat.blepluginalt I/AndroidRuntime: VM exiting with result code 1, cleanup skipped.
03-12 11:26:53.925 838-1305/? I/WindowState: WIN DEATH: Window{42acce20 u0 cl.qualitat.blepluginalt/cl.qualitat.blepluginalt.MainActivity}
03-12 11:26:53.925 838-1276/? I/ActivityManager: Process cl.qualitat.blepluginalt (pid 3317) has died.
```

jvidalgz, 2016-03-12:
Yes, that is my app.

jvidalgz, 2016-03-12:
Is there any tricky solution that I can do? maybe disable and enable the BluetoothAdapter after a while or destroy and re-create the app?

Any thoughts?

Thanks!

davidgyoung, 2016-03-12:
Actually I do see subsequent detections in the log after the app starts back up:

03-12 11:26:54.305 2062-2072/? D/BtGatt.GattService: registerClient() - UUID=3a7ba67d-bcf3-402c-a6aa-1a51a34464f5

You sure the end of this log captures a case where beacons are not detected?

Also, I see several log events related to WiFi going up and down.  Do you see your problem with WiFi disabled?