AltBeacon/android-beacon-library #350: How to transfer data between two Android devices using Beacon?

Deeptiman, 2016-03-14:
I asked one question few days back in Stackoverflow. So , i want to know can i solve my issue using android-beacon-library.

Link  : http://stackoverflow.com/questions/35937529/how-to-transfer-data-between-two-android-devices-using-beacon/

Please let me know.

zakjut, 2016-03-14:
Beacons do not really work this way. They are largely one way. Transmitting a specific frame over and over again, which receivers can pick up and perform some action depending on what they see. If you want to transmit data from device A, to device B using a bluetooth "relay" as you show in your image, beacons are not what you are looking for.

Deeptiman, 2016-03-14:
@zakjut Thank you so much , it's clear my confusion about BLE devices.

closed 2016-03-14 by davidgyoung

Francagir, 2018-02-20:
So, if you need to "get" data from a Beacon (connected to a API REST), located in a specific place(the Beacon), how could I? The moment it connects to do the query?

davidgyoung, 2018-02-20:
Simply call the API to fetch the data for the detected beacon identifier as soon as the beacon is detected

Francagir, 2018-02-20:
Thnx! And could I assign a specific location in a big place such a full city?

Francagir, 2018-02-20:
Thnx @davidgyoung!