AltBeacon/android-beacon-library #351: BootstrapNotifier does not working

Near1999, 2016-03-14:
Here is sample code that I'm using to detect enter region events:

    public class BeaconApplication extends android.app.Application implements BootstrapNotifier {
    
        private static final String TAG = "TAGTAG";
        
        @Override public void onCreate() {
            super.onCreate();
            Log.d(TAG, "App started up");
            new BackgroundPowerSaver(this);
            new RegionBootstrap(this, new Region(getPackageName(), null, null, null));
        }
    
        @Override public void didDetermineStateForRegion(int arg0, Region arg1) {
            Log.d(TAG, "didDetermineStateForRegion");
        }
    
        @Override public void didEnterRegion(Region arg0) {
            Log.d(TAG, "didEnterRegion");
        }
    
        @Override public void didExitRegion(Region arg0) {
            Log.d(TAG, "didExitRegion");
        }
    }

The problem in that if I'm using the following build.gradle configuration then everything works like expected

    compileSdkVersion 21
        buildToolsVersion "21.0.0"
        defaultConfig {
            applicationId "com.test"
            minSdkVersion 18
            targetSdkVersion 21
            versionCode 1
            versionName "1.0"
        }

LogCat:

> D/TAGTAG: App started up
> D/TAGTAG: didDetermineStateForRegion
> D/TAGTAG: didEnterRegion

But if I'm changing `compileSdkVersion ` to the actual version then nothing work

    compileSdkVersion 23
    buildToolsVersion "23.0.2"
    defaultConfig {
        applicationId "com.test"
        minSdkVersion 18
        targetSdkVersion 23
        versionCode 1
        versionName "1.0"
    }

LogCat:

> D/TAGTAG: App started up

http://stackoverflow.com/questions/35988149/altbeacon-library-background-service

davidgyoung, 2016-03-14:
Answered on StackOverflow at the link above.

closed 2016-03-14 by davidgyoung

Near1999, 2016-03-14:
@davidgyoung, I've made as You said but still have the same problem ( Nothing was changed.