AltBeacon/android-beacon-library #354: so many ServiceNotDeclaredException

cokepanm, 2016-03-16:
altbeacon version 2.7.1
compile 'org.altbeacon:android-beacon-library:2.7.1'

when i release my app for a lot app markets, i found that many devices will crash with a exception as below.
org.altbeacon.beacon.BeaconManager$ServiceNotDeclaredException
The BeaconService is not properly declared in AndroidManifest.xml. If using Eclipse, please verify that your project.properties has manifestmerger.enabled=true

I decompile my app and get AndroidManifest.xml, at the end of it i found 2 merged service decleard as below.
<service
			android:label="beacon"
			android:name="org.altbeacon.beacon.service.BeaconService"
			android:enabled="true"
			android:exported="false"
			android:isolatedProcess="false"
			>
		</service>
		<service
			android:name="org.altbeacon.beacon.BeaconIntentProcessor"
			android:enabled="true"
			android:exported="false"
			>
		</service>

I have a question that how this happen?

bbil, 2016-06-01:
+1

I've been seeing this exception as well, only from crash reports. Our manifests should be merged as well.

grocaDW, 2016-06-06:
+1
I'have been having this same issue too. Hope it gets resolved soon

davidgyoung, 2016-06-24:
Unfortunately, I don't have any theories on how to investigate this, and I can't reproduce it myself.  If anybody has a theory on why this might be happening in some cases or ideas to reproduce, please comment.

IgorGanapolsky, 2017-08-04:
I am seeing this issue right now with android-beacon-library:2.9.1.  Every time I try to call this:
`this.beaconManager = BeaconManager.getInstanceForApplication(context.getApplicationContext());`
I get this error:
> JNI DETECTED ERROR IN APPLICATION: JNI NewStringUTF called with pending exception org.altbeacon.beacon.BeaconManager$ServiceNotDeclaredException: The BeaconService is not properly declared in AndroidManifest.xml.  If using Eclipse, please verify that your project.properties has manifestmerger.enabled=true
08-04 15:27:48.465 18394-18432/air.com.crestron.andros A/art: art/runtime/java_vm_ext.cc:470]   at void org.altbeacon.beacon.BeaconManager.verifyServiceDeclaration() (BeaconManager.java:921)
08-04 15:27:48.465 18394-18432/air.com.crestron.andros A/art: art/runtime/java_vm_ext.cc:470]   at void org.altbeacon.beacon.BeaconManager.<init>(android.content.Context) (BeaconManager.java:254)
08-04 15:27:48.465 18394-18432/air.com.crestron.andros A/art: art/runtime/java_vm_ext.cc:470]   at org.altbeacon.beacon.BeaconManager org.altbeacon.beacon.BeaconManager.getInstanceForApplication(android.content.Context) (BeaconManager.java:246)

davidgyoung, 2017-08-04:
This might be triggered by some unusual project setup.  If you can share your code privately, I am happy to take a look.

IgorGanapolsky, 2017-08-04:
@davidgyoung Yes, my project setup is indeed unusual.  It uses Adobe Air, so I don't have an explicitly declared activity nor application in **AndroidManifest.xml.**

davidgyoung, 2017-08-04:
Do you have access to the Android Manistee.xml to make modifications?

IgorGanapolsky, 2017-08-04:
I do.  What do I need to look for?

davidgyoung, 2017-08-04:
Try merging the contents of this manifest into yours:

https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/AndroidManifest.xml

IgorGanapolsky, 2017-08-07:
@davidgyoung When I try to merge from your AndroidManifest into mine, I get an error from
`<service android:enabled="true"
                 ......
   <service android:name=".BeaconIntentProcessor"
                ......`

> **## Unresolved class...***

Then, I get a runtime crash:
> Process: air.com.me.andros, PID: 2910
org.altbeacon.beacon.BeaconManager$ServiceNotDeclaredException: The BeaconService is not properly declared in AndroidManifest.xml.  If using Eclipse, please verify that your project.properties has manifestmerger.enabled=true
org.altbeacon.beacon.BeaconManager.verifyServiceDeclaration(BeaconManager.java:921)

Here is what the merged manifest looks like:
`<?xml version="1.0" encoding="utf-8"?>
<manifest
    xmlns:android="http://schemas.android.com/apk/res/android"
    android:versionCode="100000000"
    android:versionName="1.00.00"
    android:installLocation="0"
    package="air.com.me.andros"
    platformBuildVersionCode="21"
    platformBuildVersionName="5.0.1-1624448">

    <application
        android:theme="@ref/0x7f0a0001"
        android:label="@ref/0x7f080000"
        android:icon="@ref/0x7f020033"
        android:name="android.support.multidex.MultiDexApplication"
        android:enabled="true"
        android:debuggable="true"
        android:allowBackup="false"
        android:hardwareAccelerated="true">

        <activity
            android:theme="@ref/0x7f0a000f"
            android:label="@ref/0x7f080000"
            android:name=".AppEntry"
            android:launchMode="2"
            android:screenOrientation="2"
            android:configChanges="0x4a0"
            android:windowSoftInputMode="0x12">

            <meta-data
                android:name="namespaceVersion"
                android:value="1102053376.000000" />

            <intent-filter>

                <action
                    android:name="android.intent.action.MAIN" />

                <category
                    android:name="android.intent.category.LAUNCHER" />
            </intent-filter>

            <intent-filter>

                <data
                    android:scheme="me"
                    android:path="launchApp" />

                <data
                    android:scheme="me-app"
                    android:path="launchApp" />

                <action
                    android:name="android.intent.action.VIEW" />

                <category
[... 215 lines cut ...]
    <uses-permission
        android:name="android.permission.VIBRATE" />

    <uses-permission
        android:name="com.android.vending.BILLING" />

    <uses-permission
        android:name="com.google.android.c2dm.permission.RECEIVE" />

    <permission
        android:name="air.com.me.andros.permission.C2D_MESSAGE"
        android:protectionLevel="0x2" />

    <uses-permission
        android:name="air.com.me.andros.permission.C2D_MESSAGE" />

    <uses-feature
        android:name="android.hardware.touchscreen.multitouch"
        android:required="true" />

    <uses-feature
        android:name="android.hardware.telephony"
        android:required="false" />
</manifest>
`

davidgyoung, 2017-08-07:
I think the problem is that the package names need to be fully qualified in
the manifest.  This is normally automatic, but since you manually copied
the manifest entries, you also must update the prefix of the package
names.  So change this:

    <service
        android:label="beacon"
        android:name=".service.BeaconService"
        android:enabled="true"
        android:exported="false"
        android:isolatedProcess="false" />

    <service
        android:name=".BeaconIntentProcessor"
        android:enabled="true"
        android:exported="false" />

To this:

    <service
        android:label="beacon"
        android:name="org.altbeacon.beacon.service.BeaconService"
        android:enabled="true"
        android:exported="false"
        android:isolatedProcess="false" />

    <service
        android:name="org.altbeacon.beacon.BeaconIntentProcessor"

        android:enabled="true"
        android:exported="false" />

IgorGanapolsky, 2017-08-07:
@davidgyoung  Your fix works now.  Thanks!