AltBeacon/android-beacon-library #356: Problem with background ranging and defining region for Eddystone

zibiksior, 2016-03-17:
Hi, I'm developing an app which provides background Beacon monitoring. I would like to start ranging when user enters beacon from defined region. Unfortunately I can't define region for Eddystone profile. When I create region object with Eddystone NamespaceID the Identifier throws parsing exception...
`Region region = new Region("backgroundRegion",Identifier.parse("f7826da6bc5b71e0893e"), null, null);`
When Identifier is in UUID format or integer according to docs ([https://altbeacon.github.io/android-beacon-library/javadoc/org/altbeacon/beacon/Identifier.html](url)) parsing works fine. So, when Beacon is set to Eddystone, NamespaceID is first Identifier which should define region exactly the same as UUID is first Identifier in iBeacon profile. Do I understand it right?  If I'm right in that situation I can't parse Eddystone NamespaceID as Identifier for region and can't monitor if user enter beacons which are set to use Eddystone profile. Please correct me if I'm wrong and please tell me how to define region for Eddystone beacons.

Ok so that's the first problem I'm facing. The second one is how to start ranging after entering a region? When app is in background and it's monitoring all the time and user enters my defined region I would like to start ranging and get InstanceID or Major, Minor value to determine what beacon is that, connect to server and send user a notfication. The best would be if I could range and communicate with server in background. I used this sample to achieve background monitoring: [https://altbeacon.github.io/android-beacon-library/samples.html](url). I also downloaded your sample project from here: [https://github.com/AltBeacon/android-beacon-library-reference](url) to base on it.

Unfortunately in this example when user enters the region Activity is launched... I don't want that to happen. Soo my question is: Is it possible to range beacons in background?

BTW I'm testing my app on Nexus 5 with Android 6.0.1

I hope you can help me with that problems!
Thank you in advance for any solutions!

davidgyoung, 2016-03-17:
@zibiksior, for help in using this library, please create questions on StackOverflow.com.  This forum is reserved for feature requests and reproducible bugs with the library itself.

Both issues you describe are easily resolved.  Please create one question for each of your two issues on StackOverflow.com, pasting in the above info and I will gladly answer there.

closed 2016-03-17 by davidgyoung

zibiksior, 2016-03-17:
Ok I will create my questions on StackOverflow. Thanks!