AltBeacon/android-beacon-library #357: Support local distance.json files and disable refresh.

awenger, 2016-03-22:
If I figured it out correctly the library refreshes the device specific distance configuration periodically via http. Is it possible to provide such a list to the library explicitly and disable the automatic update comletely? Reasons:
- The server is super slow: 37.36 Seconds (not ms)
- The data doesn't change: https://github.com/AltBeacon/android-beacon-library/issues/237
- Overall I'm not to happy about libraries that *phone home*

andersborgabiro, 2016-03-22:
That or some overall approximation would be great for offline apps. Is it a lot of data?

awenger, 2016-03-22:
no, only the config for 2 types of devices:
http://data.altbeacon.org/android-distance.json

andersborgabiro, 2016-03-22:
Sounds fair that it should be in the code, and possible to deactivate communication.

davidgyoung, 2016-03-22:
Yes, you can disable the checks for remote updates like this:

    beaconManager.setDistanceModelUpdateUrl(null);

There is already a built-in android-distance.json file bundled with the library, so you will still get the default device coefficient list if you make the change above.  It just won't do the remote update check.  

You can also replace null with a string to point to a URL of an equivalent file on a server of your own, so you can provide your own custom list of distance coefficients or just put the same list on your own server.  Unfortunately, the number of device updates to the configured list has been much smaller than I'd like.  I'm always looking for folks to help build this out.   Any volunteers? :)  I'll investigate the server performance.  No reason it should be so slow.

I'll investigate the server performance as there is no reason it should be so slow.

davidgyoung, 2016-05-03:
URL is responding in about 70 ms now.

closed 2016-05-03 by davidgyoung