AltBeacon/android-beacon-library #361: v2.8 is unstable

lubeast, 2016-04-13:
Today i found this library has 2.8 release, so i update library from v2.7 to v2.8.
[image]
And i didn't change the configuration when i update library. But you can see it's becoming very unstable and i didn't change device or beacon.
So i change v2.8 back to v2.7 it becomes normal.

[image]

My test device is `moto x pro`, system version is `5.0.2`

What may cause this problem? Or i need to change my old configuration after i update to v2.8?

davidgyoung, 2016-04-13:
Thanks for this report, @lubeast.  Can you please turn on library debugging with `beaconManager.setDebug(true)` and capture LogCat output on 2.8?

lubeast, 2016-04-13:
```
04-13 20:42:19.248 29282-29282/com.radiusnetworks.beacondecservice I/BeaconService: stop ranging received
04-13 20:42:19.248 29282-29282/com.radiusnetworks.beacondecservice D/BeaconService: Currently ranging 0 regions.
04-13 20:42:19.250 29282-29282/com.radiusnetworks.beacondecservice D/CycledLeScanner: Set scan periods called with 1100, 5000 Background mode must have changed.
04-13 20:42:19.250 29282-29282/com.radiusnetworks.beacondecservice D/CycledLeScanner: We are in the background.  Setting wakeup alarm
04-13 20:42:19.253 29282-3195/com.radiusnetworks.beacondecservice D/BeaconIntentProcessor: got an intent to process
04-13 20:42:19.253 29282-3195/com.radiusnetworks.beacondecservice D/RangingData: parsing RangingData
04-13 20:42:19.254 29282-3195/com.radiusnetworks.beacondecservice D/RangingData: parsing RangingData
04-13 20:42:19.254 29282-3195/com.radiusnetworks.beacondecservice D/BeaconIntentProcessor: got ranging data
04-13 20:42:19.254 29282-3195/com.radiusnetworks.beacondecservice D/BeaconService: beacons.size():0,com.radiusnetworks.beacondecservice.service.BeaconService$1@28b027fa
04-13 20:42:19.259 29282-29282/com.radiusnetworks.beacondecservice D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{33a5315: android.os.BinderProxy@3c63f62a}
04-13 20:42:19.286 29282-29299/com.radiusnetworks.beacondecservice D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=B4:99:4C:28:EB:4C, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[2, 21, 47, 35, 68, 84, -49, 109, 74, 15, -83, -14, -12, -111, 27, -87, -1, -90, 39, 22, -67, 22, -59]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=ebeoo-28EB4C], mRssi=-60, mTimestampNanos=191314386075758}
04-13 20:42:19.286 29282-29282/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: got record
04-13 20:42:19.287 29282-29282/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: got a filtered scan result in the background.
04-13 20:42:19.287 29282-29423/com.radiusnetworks.beacondecservice D/BeaconParser: Ignoring pdu type 01
04-13 20:42:19.287 29282-29423/com.radiusnetworks.beacondecservice D/BeaconParser: Processing pdu type FF: 0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa62716bd16c50d096562656f6f2d323845423443000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
04-13 20:42:19.287 29282-29423/com.radiusnetworks.beacondecservice D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
04-13 20:42:19.287 29282-29423/com.radiusnetworks.beacondecservice D/BeaconParser: Bytes are: 0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa62716bd16c50d096562656f6f2d323845423443000000000000000000000000000000000000
04-13 20:42:19.290 29282-29423/com.radiusnetworks.beacondecservice D/BeaconService: beacon detected : id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 10006 id3: 48406
04-13 20:42:19.290 29282-29423/com.radiusnetworks.beacondecservice D/Callback: attempting callback via intent: ComponentInfo{com.radiusnetworks.beacondecservice/org.altbeacon.beacon.BeaconIntentProcessor}
04-13 20:42:19.294 29282-29423/com.radiusnetworks.beacondecservice D/MonitoringStatus: saveMonitoringStatusIfOn()
04-13 20:42:19.297 29282-3197/com.radiusnetworks.beacondecservice D/BeaconIntentProcessor: got an intent to process
04-13 20:42:19.298 29282-3197/com.radiusnetworks.beacondecservice D/BeaconIntentProcessor: got monitoring data
04-13 20:42:19.298 29282-3197/com.radiusnetworks.beacondecservice D/BeaconIntentProcessor: Calling monitoring notifier: com.radiusnetworks.beacondecservice.service.BeaconService@94ad337
04-13 20:42:19.298 29282-3197/com.radiusnetworks.beacondecservice D/BeaconService: switch from seeing/not seeing beacons
04-13 20:42:19.298 29282-3197/com.radiusnetworks.beacondecservice D/BeaconService: didEnterRegion called
04-13 20:42:19.299 29282-3197/com.radiusnetworks.beacondecservice D/BeaconManager: This consumer is not bound.  binding: com.radiusnetworks.beacondecservice.service.BeaconService@94ad337
04-13 20:42:19.301 29282-29282/com.radiusnetworks.beacondecservice D/BeaconManager: we have a connection to the service now
04-13 20:42:19.301 29282-3197/com.radiusnetworks.beacondecservice D/BeaconManager: consumer count is now: 2
04-13 20:42:19.302 29282-29282/com.radiusnetworks.beacondecservice D/BeaconService: onBeaconServiceConnect
04-13 20:42:19.302 29282-3197/com.radiusnetworks.beacondecservice D/BeaconManager: callback packageName: com.radiusnetworks.beacondecservice
04-13 20:42:19.302 29282-29282/com.radiusnetworks.beacondecservice I/BeaconService: start ranging received
04-13 20:42:19.302 29282-29282/com.radiusnetworks.beacondecservice D/BeaconService: Currently ranging 1 regions.
04-13 20:42:19.302 29282-29282/com.radiusnetworks.beacondecservice D/CycledLeScanner: start called
04-13 20:42:19.302 29282-29282/com.radiusnetworks.beacondecservice D/CycledLeScanner: scanning already started
04-13 20:42:19.302 29282-29282/com.radiusnetworks.beacondecservice D/CycledLeScanner: Set scan periods called with 1100, 5000 Background mode must have changed.
04-13 20:42:19.302 29282-29282/com.radiusnetworks.beacondecservice D/CycledLeScanner: We are in the background.  Setting wakeup alarm
04-13 20:42:19.302 29282-29423/com.radiusnetworks.beacondecservice D/BeaconService: looking for ranging region matches for this beacon
[... cut ...]
04-13 20:42:31.461 29282-3554/com.radiusnetworks.beacondecservice D/BeaconIntentProcessor: Calling monitoring notifier: com.radiusnetworks.beacondecservice.service.BeaconService@94ad337
04-13 20:42:31.461 29282-3554/com.radiusnetworks.beacondecservice D/BeaconService: switch from seeing/not seeing beacons
04-13 20:42:31.461 29282-3554/com.radiusnetworks.beacondecservice D/BeaconService: didExitRegion called
04-13 20:42:31.461 29282-3554/com.radiusnetworks.beacondecservice D/BeaconManager: callback packageName: com.radiusnetworks.beacondecservice
```

These are between one start ranging and stop ranging.

davidgyoung, 2016-04-13:
Based on these logs, I see the app is in the background, and the library is configured to do a full scan to look for beacons for 1100ms every 5000 ms, deferring to hardware filters the rest of the time.  There was a detection at 20:42:20.239, which was triggered by a hardware filter.  Because beacons were in the vicinity, hardware filters were then temporarily disabled.  So the library spent 1100ms scanning for beacons at:  20:42:24.211 and  then again at 20:42:30.302 but none were seen.  The library then fired a region exit at  20:42:31.461 because no beacons had been detected in 10 seconds.

Does this sound like incorrect behavior?  Do you believe the beacon was transmitting the whole time?  I don't see any OS-level log messages for bluetooth advertisements being received, suggesting Android is really not seeing them.

lubeast, 2016-04-14:
Sorry for the delay. Yes your analyse is right. But i do believe the beacon was transmitting the whole time. Because 2.7 can detect beacon every single scan-cycle. That's the reason why i doubt 2.8 is unstable. And i will try to figure out why. Thanks for your answer @davidgyoung :D

davidgyoung, 2016-04-14:
Can you attach a log for the same conditions on 2.7?  It may give a clue.

lubeast, 2016-04-14:
```
04-14 09:53:54.102 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScanner: starting a new scan cycle
04-14 09:53:54.103 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScanner: starting a new bluetooth le scan
04-14 09:53:54.103 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: starting non-filtered scan in SCAN_MODE_LOW_LATENCY
04-14 09:53:54.107 31748-31770/com.radiusnetworks.beacondecservice D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
04-14 09:53:54.108 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
04-14 09:53:54.117 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{13d69428: android.os.BinderProxy@12717541}
04-14 09:53:54.117 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScanner: Scan started
04-14 09:53:54.377 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: got record
04-14 09:53:54.377 31748-31896/com.radiusnetworks.beacondecservice D/BeaconParser: Ignoring pdu type 01
04-14 09:53:54.377 31748-31896/com.radiusnetworks.beacondecservice D/BeaconParser: Processing pdu type FF: 0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa62716bd16c50d096562656f6f2d323845423443000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
04-14 09:53:54.378 31748-31896/com.radiusnetworks.beacondecservice D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
04-14 09:53:54.379 31748-31896/com.radiusnetworks.beacondecservice D/BeaconService: beacon detected : id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 10006 id3: 48406
04-14 09:53:54.380 31748-31896/com.radiusnetworks.beacondecservice D/BeaconService: looking for ranging region matches for this beacon
04-14 09:53:54.380 31748-31896/com.radiusnetworks.beacondecservice D/BeaconService: matches ranging region: id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: null id3: null
04-14 09:53:54.380 31748-31896/com.radiusnetworks.beacondecservice D/RangeState: adding id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 10006 id3: 48406 to existing range for: org.altbeacon.beacon.service.RangedBeacon@afe8b79
04-14 09:53:54.479 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: got record
04-14 09:53:54.479 31748-31899/com.radiusnetworks.beacondecservice D/BeaconParser: Ignoring pdu type 01
04-14 09:53:54.480 31748-31899/com.radiusnetworks.beacondecservice D/BeaconParser: Processing pdu type FF: 0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa62716bd16c50d096562656f6f2d323845423443000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
04-14 09:53:54.480 31748-31899/com.radiusnetworks.beacondecservice D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
04-14 09:53:54.483 31748-31899/com.radiusnetworks.beacondecservice D/BeaconService: beacon detected : id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 10006 id3: 48406
04-14 09:53:54.483 31748-31899/com.radiusnetworks.beacondecservice D/BeaconService: looking for ranging region matches for this beacon
04-14 09:53:54.483 31748-31899/com.radiusnetworks.beacondecservice D/BeaconService: matches ranging region: id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: null id3: null
04-14 09:53:54.484 31748-31899/com.radiusnetworks.beacondecservice D/RangeState: adding id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 10006 id3: 48406 to existing range for: org.altbeacon.beacon.service.RangedBeacon@afe8b79
04-14 09:53:54.579 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: got record
04-14 09:53:54.580 31748-31908/com.radiusnetworks.beacondecservice D/BeaconParser: Ignoring pdu type 01
04-14 09:53:54.580 31748-31908/com.radiusnetworks.beacondecservice D/BeaconParser: Processing pdu type FF: 0201061aff4c0002152f234454cf6d4a0fadf2f4911ba9ffa62716bd16c50d096562656f6f2d323845423443000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
04-14 09:53:54.580 31748-31908/com.radiusnetworks.beacondecservice D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
04-14 09:53:54.582 31748-31908/com.radiusnetworks.beacondecservice D/BeaconService: beacon detected : id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 10006 id3: 48406
04-14 09:53:54.582 31748-31908/com.radiusnetworks.beacondecservice D/BeaconService: looking for ranging region matches for this beacon
04-14 09:53:54.583 31748-31908/com.radiusnetworks.beacondecservice D/BeaconService: matches ranging region: id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: null id3: null
04-14 09:53:54.583 31748-31908/com.radiusnetworks.beacondecservice D/RangeState: adding id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: 10006 id3: 48406 to existing range for: org.altbeacon.beacon.service.RangedBeacon@afe8b79
04-14 09:53:54.639 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: got record
04-14 09:53:54.639 31748-32004/com.radiusnetworks.beacondecservice D/BeaconParser: Ignoring pdu type 01
04-14 09:53:54.639 31748-32004/com.radiusnetworks.beacondecservice D/BeaconParser: Processing pdu type FF: 02010613ff4c000c0e007e3dab76491bef2635e3c16e97000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 22
[... cut ...]
04-14 09:54:03.378 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 2923 milliseconds
04-14 09:54:04.379 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 1922 milliseconds
04-14 09:54:05.380 31748-31748/com.radiusnetworks.beacondecservice D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 921 milliseconds
```
These are 2.7 logs

davidgyoung, 2016-04-14:
Thanks.  I do see an immediate detection after resuming low latency scanning in the background.  Let me run some tests to see if I can reproduce.

lubeast, 2016-04-14:
Yes, i saw this difference at the top of 2.7 log. Thanks for your help @davidgyoung

davidgyoung, 2016-04-14:
@lubeast, I tried to reproduce this with the library's reference app, running it on a Nexus 5X with the 2.8 version of the library.  I gave it the following customizations:

```
beaconManager.setDebug(true);
beaconManager.setBackgroundBetweenScanPeriod(30000l);
beaconManager.setBackgroundScanPeriod(1100l);
```

I then turned on a single AltBeacon and let the app run in the background.  It continued detecting the beacon regularly, and there were no exit events recorded.

Unfortunately, this means I can't reproduce your issue.   It may very well be a difference between 2.7 and 2.8, but if so it sounds like it is also specific to your app or your phone.   If you are interested in troubleshooting further, you might try the reference app and see if it gives you the same trouble. https://github.com/altbeacon/android-beacon-library-reference

lubeast, 2016-04-15:
@davidgyoung ok, I'll try the reference app. Thanks again for your help.

closed 2016-04-15 by lubeast