AltBeacon/android-beacon-library #362: not detecting TI senor tag

niccs, 2016-04-14:
###Expected behavior

###Actual behavior

###Steps to reproduce this behavior

###Mobile device model and OS version -- Nexus 5, API 23

###Android Beacon Library version -- 2.3.5

I have a sensor tag from TI(Texas Instruments), and the firmware is TI provided.  The same 2541 chipset is used in all beacons.  I try to detect it using this library, it comes inside the didRangeBeaconsInRegion() callback but it prints beacons.size to be zero size.

Now, what changes I need to do, to get this chipset detected?

Just for further information, there are free apps in playstore named (iBeacon detector) which detects this chipset.

Any help will be highly appreciated.

Thank You

andersborgabiro, 2016-04-14:
But is it running code to behave like a beacon of some type (iBeacon, Eddystone...)?

It might be detected as a BLE device, but that's not the same thing.

niccs, 2016-04-14:
Thanks for the quick response.

I understand it, but what modifications can I do, to get it detected?  
Will implementing beaconParser  can be a solution?

andersborgabiro, 2016-04-14:
The sensor tag needs to act as a beacon. It's likely nothing wrong on the phone side.

Did you do this? http://processors.wiki.ti.com/index.php/SensorTag_with_iBeacon

niccs, 2016-04-14:
I understand, that is one way, but instead of changing its firmware of sensor tag  I would preferrably get beacons.
Actually my requirement is to use this library to detect a BLE device, which existing apps are able to do. So where should I make changes ,so that I can benefit from the design of this library.

andersborgabiro, 2016-04-14:
I guess David should respond, but this library is not for detecting BLE devices in general, rather specifically beacons, which is a special case (beacons send broadcast messages continuously).

In other words, an app using this library would get information about beacons or nothing. You use the BLE features of Android to access BLE devices in a general sense. I've done it, therefore it can be done :).

davidgyoung, 2016-04-14:
@andersborgabiro is correct.  I have put up a more detailed response here: 
http://stackoverflow.com/questions/36620303/android-beacon-library-version-2-3-5-not-detecting-ti-senor-tag/36622985#36622985

closed 2016-04-14 by davidgyoung