AltBeacon/android-beacon-library #363: checkLocationPermission no exclude less than api23 in CycledLeScanner.java file

blancell, 2016-04-15:
###Expected behavior
Can detect beacon device
###Actual behavior
Don't detect beacon device
###Steps to reproduce this behavior
device android os api less than 23 
###Mobile device model and OS version
android 4.3 
###Android Beacon Library version
2.8

davidgyoung, 2016-04-15:
@blancell, I can not reproduce this issue using the reference application.  I changed the build.gradle file to set `targetSdkVersion 19` and ran the application with an AltBeacon in the vicinitiy.  The beacon properly triggered the reference app to enter and exit the monitored region and to get ranging callbacks.

If you can provide more specifics on the issue you are seeing it may be helpful.

blancell, 2016-04-16:
@davidgyoung,  my  test android device sdk api 18. targetSdkVersion 23 in build.gradle. 
try {
           if (checkLocationPermission()) {
                  startScan();
           }
 } catch (Exception e) {
          LogManager.e(e, TAG, "Internal Android exception scanning for beacons");
}
in the code when check location permission less than 23, return false don't start scan?
I modify checkLocationPermission method, can detect device now.
Thanks!

davidgyoung, 2016-04-16:
Sorry, I'm not sure I understand the issue.  The `checkLocationPermission()` just verifies that there is a location permission in the AndroidManifest.xml (which the library supplies and will merge into the app manifest automatically.)  It also checks to see if the permission has been granted, which should be automatic pre-SDK 23.  So which is causing the problem?  (1) Do you not have the location permission entries in the manifest for your app? or (2) Is your device reporting that permission has not been granted.

I want to make sure I fully understand so I can put in the proper fix.

```
    private boolean checkLocationPermission() {
        return checkPermission(Manifest.permission.ACCESS_COARSE_LOCATION) || checkPermission(Manifest.permission.ACCESS_FINE_LOCATION);
    }

    private boolean checkPermission(final String permission) {
        return mContext.checkPermission(permission, android.os.Process.myPid(), android.os.Process.myUid()) == PackageManager.PERMISSION_GRANTED;
    }

```

davidgyoung, 2016-04-17:
@blancell, can you please add the following code to your app's `onCreate` method of your main activity, and let me know what it outputs in the log?  You should see something similar to what I show below.

```
        try {
            PackageInfo info = getPackageManager().getPackageInfo(this.getPackageName(), PackageManager.GET_PERMISSIONS);
            Log.d(TAG, "SDK "+Build.VERSION.SDK_INT+" App Permissions:");
            if (info.requestedPermissions != null) {
                for (String p : info.requestedPermissions) {
                    int grantResult = this.checkPermission(p, android.os.Process.myPid(), android.os.Process.myUid());
                    if (grantResult == PackageManager.PERMISSION_GRANTED) {
                        Log.d(TAG, p+" PERMISSION_GRANTED");
                    }
                    else {
                        Log.d(TAG, p+" PERMISSION_DENIED: "+grantResult);
                    }
                }
            }
        } catch (Exception e) {
            Log.d(TAG, "Cannot get permissions due to error", e);
        }
```

Here's the output I see:

```
04-17 12:04:01.943 30624-30624/? D/MonitoringActivity: SDK 23 App Permissions:
04-17 12:04:01.944 30624-30624/? D/MonitoringActivity: android.permission.INTERNET PERMISSION_GRANTED
04-17 12:04:01.945 30624-30624/? D/MonitoringActivity: android.permission.BLUETOOTH PERMISSION_GRANTED
04-17 12:04:01.945 30624-30624/? D/MonitoringActivity: android.permission.BLUETOOTH_ADMIN PERMISSION_GRANTED
04-17 12:04:01.946 30624-30624/? D/MonitoringActivity: android.permission.RECEIVE_BOOT_COMPLETED PERMISSION_GRANTED
04-17 12:04:01.946 30624-30624/? D/MonitoringActivity: android.permission.ACCESS_COARSE_LOCATION PERMISSION_GRANTED

```

Seeing the output you get will help me put in a proper fix for this problem.

blancell, 2016-04-18:
@davidgyoung ,  here's the output: 
04-18 08:39:08.130 6994-6994/com.tsingche.app.tsingbeacon E/MainActivity: SDK 18 App Permissions:
04-18 08:39:08.130 6994-6994/com.tsingche.app.tsingbeacon E/MainActivity: android.permission.BLUETOOTH PERMISSION_GRANTED
04-18 08:39:08.130 6994-6994/com.tsingche.app.tsingbeacon E/MainActivity: android.permission.BLUETOOTH_ADMIN PERMISSION_GRANTED
04-18 08:39:08.130 6994-6994/com.tsingche.app.tsingbeacon E/MainActivity: android.permission.RECEIVE_BOOT_COMPLETED PERMISSION_GRANTED

I can't see ACCESS_COARSE_LOCATION, but my sdk 18.
<uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"/>  in the library. 
Whether add ACCESS_COARSE_LOCATION permission to every sdk ? 
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
I add ACCESS_COARSE_LOCATION to app, can detect device now.
My mistake ，thanks !

davidgyoung, 2016-04-18:
@blancell, glad that solved it for you, but it may not be your mistake.  

The library has its own AndroidManifest.xml that includes ACCESS_COARSE_LOCATION.  It is supposed to be automatically merged into the manifest if your app, but it sounds like it is not.  This concerns me.

How are you building your project?  Android studio or Eclipse?  Did you manually add those other permissions into your project's manifest?

blancell, 2016-04-18:
I use Android Studio.
Yes, i manually add  ACCESS_COARSE_LOCATION to my app manifest.
May be uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION" use in sdk 23 only?

davidgyoung, 2016-04-18:
That's it!  Thanks, @blancell, for helping me find a critical bug.  Yes, the best solution is to conditionally go ahead and scan if the device's SDK is pre 23, even if location is not granted.

davidgyoung, 2016-04-18:
Fixed in #365

closed 2016-05-03 by davidgyoung