AltBeacon/android-beacon-library #366: add/remove regions from RegionBootstrap

aristide, 2016-04-24:
Hi @davidgyoung  !! is there any way of add or remove a monitored region from a running RegionBoostrap class. 

I saw your [answer](http://stackoverflow.com/a/30099289) where you suggested to use beaconManager. But  nn RegionBootstrap.java, i was looking for methods like
**addRegsion(Region region)/ removeRegion(Region region)** that can ease adding and removing operations.

thanks!!

davidgyoung, 2016-04-24:
Understood.  I'll mark this as a feature request as it seems like a reasonable ask.  If you'd like to take a short at implementing it yourself, I'd welcome a pull request.

aristide, 2016-04-25:
@davidgyoung 
my pull request #368

davidgyoung, 2016-07-22:
Implemented in #368

closed 2016-07-22 by davidgyoung