AltBeacon/android-beacon-library #371: Large amount of errors from bt-btm when monitoring starts.

zakjut, 2016-04-26:
### Behaviour

I have noticed a humongous amount of errors from 'bt-btm' when I start monitoring for beacons. Very soon after calling startMonitoringBeaconsInRegion, dozens of errors with bt-btm new address ab:cd:ef:gh:ij:kl start popping up. It doesn't seem to matter if there is a beacon actually broadcasting or not. The service still works correctly, both monitoring and ranging but I am curious what this could be about.

04-25 15:50:15.029 D/LAACService(24736): Beacon Service connected, starting monitoring.

04-25 15:50:15.048 D/BtGatt.GattService( 7602): registerClient() - UUID=7183e20d-2397-49e9-8b9d-11bf054f0645

04-25 15:50:15.049 D/BtGatt.GattService( 7602): onClientRegistered() - UUID=7183e20d-2397-49e9-8b9d-11bf054f0645, clientIf=9

04-25 15:50:15.049 D/BluetoothLeScanner(24736): onClientRegistered() - status=0 clientIf=9

04-25 15:50:15.050 D/BtGatt.GattService( 7602): start scan with filters

04-25 15:50:15.052 D/BtGatt.ScanManager( 7602): handling starting scan

04-25 15:50:15.059 I/WCNSS_FILTER( 7923): ibs_msm_serial_clock_vote: vote UART CLK ON using UART driver's ioctl()

04-25 15:50:15.065 D/BtGatt.GattService( 7602): onScanFilterEnableDisabled() - clientIf=9, status=0, action=1

04-25 15:50:15.065 D/BtGatt.ScanManager( 7602): callback done for clientIf - 9 status - 0

04-25 15:50:15.067 D/BtGatt.GattService( 7602): onScanFilterParamsConfigured() - clientIf=9, status=0, action=0, availableSpace=15

04-25 15:50:15.067 D/BtGatt.ScanManager( 7602): callback done for clientIf - 9 status - 0

04-25 15:50:15.067 D/BtGatt.ScanManager( 7602): configureRegularScanParams() - queue=1

04-25 15:50:15.067 D/BtGatt.ScanManager( 7602): configureRegularScanParams() - ScanSetting Scan mode=2 mLastConfiguredScanSetting=-2147483648

04-25 15:50:15.133 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:15.133 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:15.319 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:15.320 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:15.431 E/NetlinkEvent( 462): NetlinkEvent::FindParam(): Parameter 'ALERT_NAME' not found

04-25 15:50:15.431 E/NetlinkEvent( 372): NetlinkEvent::FindParam(): Parameter 'ALERT_NAME' not found

04-25 15:50:15.437 D/HeadsetStateMachine( 7602): Disconnected process message: 10, size: 0

04-25 15:50:15.441 D/KeyguardUpdateMonitor( 4259): received broadcast android.intent.action.BATTERY_CHANGED

04-25 15:50:15.455 D/KeyguardUpdateMonitor( 4259): handleBatteryUpdate

04-25 15:50:15.474 D/TaskPersister( 1101): removeObsoleteFile: deleting file=3580_task.xml

04-25 15:50:15.500 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:15.500 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:15.683 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:15.684 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:15.864 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:15.864 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:16.049 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:16.050 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:16.089 I/WCNSS_FILTER( 7923): ibs_msm_serial_clock_vote: vote UART CLK OFF using UART driver's ioctl()

04-25 15:50:16.156 D/BtGatt.GattService( 7602): stopScan() - queue size =1

[... 97 lines cut ...]

04-25 15:50:17.325 D/BtGatt.GattService( 7602): onScanFilterEnableDisabled() - clientIf=9, status=0, action=1

04-25 15:50:17.325 D/BtGatt.ScanManager( 7602): callback done for clientIf - 9 status - 0

04-25 15:50:17.327 D/BtGatt.GattService( 7602): onScanFilterParamsConfigured() - clientIf=9, status=0, action=0, availableSpace=15

04-25 15:50:17.327 D/BtGatt.ScanManager( 7602): callback done for clientIf - 9 status - 0

04-25 15:50:17.327 D/BtGatt.ScanManager( 7602): configureRegularScanParams() - queue=1

04-25 15:50:17.327 D/BtGatt.ScanManager( 7602): configureRegularScanParams() - ScanSetting Scan mode=2 mLastConfiguredScanSetting=-2147483648

04-25 15:50:17.334 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

04-25 15:50:17.334 E/bt-btm ( 7602): new address: 50:2e:b5:8d:c6:dc

### Steps to reproduce this behavior
Start monitoring for beacons.

### Mobile device model and OS version
OnePlus 2 with Oxygen OS 2.2.1 (Android 5.1.1)

### Android Beacon Library version
At first I noticed the possible issue on 2.7 due to using the Xamarin binding version of the library, but I have now double checked that it occurs on the actual Android version as well (which uses the latest version of the library.)

davidgyoung, 2016-04-26:
The android bluetooth stack is notoriously verbose with their logging.  You can see the source code that generates this log message here:  https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/lollipop-wear-release/stack/btm/btm_ble_gap.c#2440

It doesn't look like it is actually an error condition, but for some reason it is set to an error level log message, perhaps incorrectly.  I suspect it is safe to ignore.

Since this is part of the Android bluetooth stack, it is out of scope of the changes that can be made with this library.  You might want to open a ticket with Google about the level of logging, which is certainly annoying.  If you have evidence that these log lines are an indication of a problem that the library might be causing, please comment and reopen this issue.

closed 2016-04-26 by davidgyoung

zakjut, 2016-04-27:
Ah, nothing to worry about then. Thanks for looking into it.