AltBeacon/android-beacon-library #372: Fatal Exception from Library

kpbird, 2016-04-28:
Fatal Exception: java.lang.UnsupportedOperationException: Only supported for Identifiers with max byte length of 2
       at org.altbeacon.beacon.Identifier.toInt(Identifier.java:254)
       at no.npro.mittakerbrygge.service.BeaconService.didRangeBeaconsInRegion(BeaconService.java:377)
       at org.altbeacon.beacon.BeaconIntentProcessor.onHandleIntent(BeaconIntentProcessor.java:65)
       at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:65)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:145)
       at android.os.HandlerThread.run(HandlerThread.java:61

###Expected behavior

###Actual behavior

###Steps to reproduce this behavior

###Mobile device model and OS version

###Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2016-05-03:
As the exception describes, it sounds like you made a call to the `.toInt()` method of identifier for an identifier that is not a two byte value.  Perhaps you have the library configured to look for an Eddystone or other beacon with a second identifier that is more than two bytes?  If so, you should not call this method.

kpbird, 2016-05-06:
@davidgyoung Thanks for the response. Yes I am calling .toInt() method. I configured iBeacon only
m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24

davidgyoung, 2016-05-06:
Can you show the code where you care calling .toInt()?  I would expect you to get that exception only if calling the method on the first beacon identifier, which is 16 bytes long, so cannot be converted to an integer.

closed 2016-06-24 by davidgyoung

lassana, 2018-08-18:
For future googlers: make sure your handle the `id2` and `id3` as major and minor vaue of the iBeacon but **not `id1`**.