AltBeacon/android-beacon-library #374: Problems on decoding EDDYSTONE_URL

mario-pucci, 2016-04-29:
Hi all,
i'm using the new 2.8 library for my app: first step is decoding a EDDYSTONE-URL frame.
Everything works great but i found that the url decoded miss the last char: in my beacon i've put a url in the form https://xxxxx.beecon.it but i decode https://xxxxx.beecon.i missing the last "t" char.

I checked your definition of BeaconParser (both from code and examples) and have: 
EDDYSTONE_URL_LAYOUT = "s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-20v";
Looking at Eddystone specification the frame can have a length 3-20 and adding the 2 bytes of the service UUID (=feaa) we get a total length of 3-22 so bytes in the frame are indexed from 0 to 21

With this in mind the correct parser layout should be
"s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-21v"
and all my decodings are correct.

Then a question: why the value of p pattern is prefixed by colon while m pattern value is prefixed by equal? Shouldn't be better to have p:x-y=P insted of p:x-y:P?

Thanks

MP

davidgyoung, 2016-04-29:
According to the spec, the max length of the encoded  URL  is 17 bytes.  See here:

https://github.com/google/eddystone/blob/master/eddystone-url/README.md

The layout expression segment Ii:4-20v ", the offsets are inclusive, so the max length is 17 bytes.  I believe this is correct.

Can you share the exact URL you are broadcasting?  Are you sure its encoded identifier has 17or fewer bytes?

To answer your second question, the reason the p term does not use an = sign before the -41 is because it is not a comparison.  This is a value that is applied to the parsed power field to get a 1m measured power value.  For Eddystone it is needed to subtract 41dB from its 0m calibration distance (as noted in the spec) to get a 1m calibration distance.

mario-pucci, 2016-04-29:
David,
you are rigth about url length but forgot 1 byte; the URL Schema that is between power and url. The WHOLE url is so 18byte

M

davidgyoung, 2016-04-29:
Ah, good point.   Yes, since this library combines the URL Schema field with the URL field to make a single identifier field, the length should be 18 bytes as you say.

closed 2016-05-02 by mario-pucci

locly, 2016-06-17:
We've just been caught out by this. The documentation on the [eddystone how-to page](https://altbeacon.github.io/android-beacon-library/eddystone-how-to.html) still has the incorrect layout. Would it be possible to get this updated?

```
- s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-20v
+ s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-21v
```

davidgyoung, 2016-06-17:
Thanks for the reminder.  I have updated this now.