AltBeacon/android-beacon-library #376: Nexus 5X Distance coefficients

sbrkx, 2016-04-30:
Ratio;Distance
0.628289047089257;0.25
0.711240435741149;0.5
0.909715317703905;1
1.02927437347653;2
1.06590983909101;3
1.09049715074721;4
1.20105870008606;5
0.992541612254315;6
1.09269957873336;7
1.09310690810803;8
1.02758192329481;9
1.06406432369342;10
1.05981354198187;12
1.17457435574411;14
1.15891111870881;16
1.07676581201797;18
1.10232135909739;20
1.20931137336273;25
1.21903773829594;30

LG Nexus 5X (Android 6.0.1)
intercept	-0.05377179
multiplier	5.46638125
power	6.98755661 

Beacon:
Estimote 100 ms advertising interval 
Transmit power: -8dbm
Measured power: -68dbm

Are these values correct?

TomEB, 2016-11-21:
What is the status on the Nexus 5X coefficients?
The current readouts on the Nexus 5X are highly inaccurate.

davidgyoung, 2016-11-22:
@TomEB, it would be helpful if you could provide:

* The make and mode of the beacon you are testing with.
* What its calibrated power setting is configured to 
* What your Nexus 5X measures as its RSSI on average at 1 m
* What this library estimates as its distance at 1 m