AltBeacon/android-beacon-library #378: BeaconManager.bind in a non activity class / service

Lugo239, 2016-05-02:
Greetings!

I am using the org.altbeacon:android-beacon-library:2.7.1 for my android application. 
Is there way to range beacons in a non activity class? 
I want to write a BeaconReceiver (implements BeaconConsumer) for range beacons in region. I tested my code in my MainActivity, but now i need this in a seperate class.

The error appears here in the constructor:  
`beaconManager.bind(this);`

Error:
`Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String android.content.Context.getPackageName()' on a null object reference`

Maybe i need to use a service. Is there any example?

Kind regards!

davidgyoung, 2016-05-02:
Please see this answer here:  http://stackoverflow.com/a/32866373/1461050

If you need more help, please open a question on Stackoverflow.com as this forum is reserved for bug reports and feature requests with the library itself.

closed 2016-05-02 by davidgyoung

Lugo239, 2016-05-03:
Everything is working! Thanks a lot for this fast and high-value answer.
(Did not find this post already, my bad)