AltBeacon/android-beacon-library #379: 2.7 to 2.8 can't detect after first launch

MartySGL, 2016-05-02:
Hi,

Since I updated Altbeacon to 2.8, the beacon detection stops once the app is killed, and never start again (even if I launch the app again it doesn't detect beacons anymore until I uninstall and re-install the app).
If I switch back to 2.7, everything works fine.

I have a LG G3 (D855) on Android 6.0

I can't understand why ?

davidgyoung, 2016-05-02:
The main change in 2.8 is that the state of monitored regions is persisted across restarts of the scanning service.  This means if you are "in" a region when you stop your app, when you start it you won't get a new"didEnterRegion" callback until the library detects you exiting that region.  This is intended to eliminate multiple "didEnterRegion" callbacks as the app is repeatedly stopped by the OS and restarted due to low memory state.

Do you think this describes what you are seeing?

MartySGL, 2016-05-03:
Yes I think, but how should I do if I never exit the region ?

davidgyoung, 2016-05-03:
If you want to guarantee a new entry event after app launch, you can add code to call `beaconManager.stopMonitoringBeaconsInRegion(region)` when your app first launches, before you make the call to start monitoring.

ovy9086, 2016-06-01:
@davidgyoung what do you mean by:

> [...]the state of monitored regions is persisted across restarts of the scanning service. This means if you are "in" a region when you stop your app, when you start it [...]
Are they somewhere persisted in `SharedPreferences` or `SQLite` ? Because otherwise, after an app stop (eg: swipe from recents) they will be cleared from statics or whatever.

davidgyoung, 2016-06-01:
Yes, it is persisted in a file store.  This has tripped up several people so I'd like to make this easy to configure to work one way or another so apps get the benefit of not getting repeated `didEnterRegion` calls across app restarts due to low memory conditions, but in development it doesn't surprise people like you.  Any ideas are welcome.

davidgyoung, 2016-07-22:
Fixed in #405

closed 2016-07-22 by davidgyoung

davidgyoung, 2016-07-25:
Released in 2.9