AltBeacon/android-beacon-library #380: Beacon monitoring service killed by Samsung devices

FrancoSabadini, 2016-05-09:
I'm having an issue on some Samsung devices, on which the beacons service stops running because the OS kills it, and it isn't restarted even though it returns `START_STICKY` from `onStartCommand()`.

Not sure if this is something you can create a workaround for, but wanted to let you know in case you were not aware of it, and also to see if you have a suggestion on what to do.

###Steps to reproduce this behavior
1. Start app that uses the beacons library.
2. Put the app in background with HOME key.
3. Verify the beacon service is running.
4. Turn off the screen by pressing the power button. 
5. Wait 10 minutes (sometimes even more, maybe a couple of hours) and turn on the screen again. 
6. Check if the beacons service is running.

###Expected behavior
The beacons service is still running in the background and scanning for beacons.

###Actual behavior
The beacons service has beeing killed and is not restarted (even though `START_STICKY` is returned from `onStartCommand()`).

###Mobile device model and OS version
Samsung Galaxy S6 Edge Android 5.1.1

###Android Beacon Library version
2.6.1

davidgyoung, 2016-05-09:
Any chance you could turn on low-level debug logging with `beaconManager.setDebug(true)`, then reproduce this issue while capturing the LogCat output?

FrancoSabadini, 2016-05-09:
Sure thing @davidgyoung, will try to do that this week and upload the logs here.

LudySu, 2016-05-10:
Hi @davidgyoung , I am working with @FrancoSabadini to look into this issue.

**Device**: Samsung S6 Edge with Android 5.1.1
**Test app**: co.vmob.sample, has set `beaconManager.setDebug(true)`. PID was 15454.

You can see from the attached log file that the app was killed by Samsung when the screen was turned off:

> 05-10 16:25:36.175 3506 3529 I ActivityManager: Force stopping service ServiceRecord
    Unknown macro: {2206ead u0 co.vmob.sample/co.vmob.sdk.location.beacon.BeaconScannerService}
> 05-10 16:25:36.175 3506 3529 V ActivityManager_GCMKill: -------- New Kill Target package : co.vmob.sample

As far as we investigate, this is because of Samsung's modified version of ActivityManager, if there are multiple apps running in the background Samsung might kill some of the them, probably to save battery. The more apps running in the background, the more chances the apps will get killed. This affect all kinds of apps, not only the ones that monitoring beacons, as we can see other apps being killed by Samsung as well.

While on a Nexus 6 or Samsung S6 with Android 6, this issue does not happen. Also tested previously on Samsung S6/Note 4 with Android 5.1.1, all had the same issue.

[beacon_service_killed.txt](https://github.com/AltBeacon/android-beacon-library/files/256384/beacon_service_killed.txt)

davidgyoung, 2016-05-11:
The library uses the `AlarmManager` to automatically re-launch the app within 5 minutes of it being killed for this exact case.  This allows the phone a few minutes to run memory intensive tasks before restarting scanning.  This mechanism is only active when the library is in background mode, because that is the only time the OS should be killing the app due to a low memory condition (which is exactly what is happening in this case.)

Reviewing the log, it appears that the device is never put into background mode with `beaconManager.setBackgroundMode(true);`  This is done automatically if using the `BackgroundPowerSaver`.  Can you verify if you code is either using the `BackgroundPowerSaver` or calling `beaconManager.setBackgroundMode(true);` ?  If not, you simply need to do so.  An appropriate time to make this call would be when the scan periods are set, as shown here in the logs:

```
05-10 16:25:25.035 24489 24489 D CycledLeScanner: Set scan periods called with 1100, 2000 Background mode must have changed.
05-10 16:25:25.035 24489 24489 D CycledLeScanner: We are not in the background.  Cancelling wakeup alarm
```

FrancoSabadini, 2016-05-11:
Thanks for the info @davidgyoung :), we'll check if we are using this and if we are not we'll try it and let you know if it worked.

LudySu, 2016-05-11:
Hi @davidgyoung , in the source file _BackgroundPowerSaver.java_ I see it requires Android API level to be >= 18, but `Application.ActivityLifecycleCallbacks` interface only requires >= 14, it there any other specific reason to require a higher API level? 

Thanks.

davidgyoung, 2016-05-11:
No, there is no good reason for requiring API 18.  

The only reason I made it 18 and not 14 is because it seemed a little silly to allow it to be used with API 14, when there is no point in using the class at all unless you have API 18 -- it won't be able to save battery by delaying bluetooth scans because you can't do Bluetooth LE scans without API 18.  But usefulness isn't really the point of declaring required API levels, so I it really should require only 14.

FrancoSabadini, 2016-05-12:
@davidgyoung our initial tests using `BackgroundPowerSaver` are very promising, the issue seems to have been fixed with this. I will let you know in more certainty once we implement it fully and our testing team has a run at it.

Thanks for the help!!

LudySu, 2016-05-13:
Hi @davidgyoung , I would like to add some more information.

When we encountered this issue earlier this year, we tried using `AlarmManager` to restart our beacon service after 15mins when being killed. But that was not successful because we found Samsung killed our beacon service again when it tried to restart.

We guess this is because Samsung has tuned their `ActivityManger` a little bit since then, so it become less aggressive in killing background apps. On our previous test, the app will always get killed by Samsung when screen is turned off; but now only when there are multiple apps in the background and Samsung will only kill the least recent launched ones.

Again thanks for your help.

closed 2016-05-31 by davidgyoung