AltBeacon/android-beacon-library #382: didEnterRegion with RegionBootstrap fire whereas I don't move from the Beacon

kmenager, 2016-06-09:
Hi,

I had an issue with RegionBootstrap. I would like to scan for specific beacon and only trigger new beacon in my area. But when I stay in beacon region, I have the method didEnterRegion triggered randomly.

Any idea how to prevent this ?

Here is my code in Application class
```java
    @Override
    public void onCreate() {
        super.onCreate();
        mBeaconManager = BeaconManager.getInstanceForApplication(this);
        mBeaconManager.getBeaconParsers().add(
                new BeaconParser()
                     .setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));

        backgroundPowerSaver = new BackgroundPowerSaver(this);
        mRegion = new Region("uniqueId", Identifier.fromUuid(UUID), null, null);
        regionBootstrap = new RegionBootstrap(this, mRegion);

        mBeaconManager.setBackgroundBetweenScanPeriod(10000);
    }
```

davidgyoung, 2016-06-11:
For assistance in using this library, please open a question on StackOverflow.com.  This forum is reserved for feature requests and reproducible bug reports with the library itself.

closed 2016-06-11 by davidgyoung

kmenager, 2016-06-14:
@davidgyoung I don't require assistance. I noticed in background mode, that every time the service comes alive, it fire new didEnterRegion while I don't move from the region.
I thinks it's an issue and could be simple reproducible.

davidgyoung, 2016-06-14:
@kmenager, I believe this is resolved as of library version 2.8 in this pull request:  https://github.com/AltBeacon/android-beacon-library/pull/315