AltBeacon/android-beacon-library #383: 2.8.1 causes app to freeze when adding Regions to monitor

marwfair, 2016-06-12:
When creating several hundred Regions and starting to monitor them, the app freezes for a minute, usually longer. This is happening every time and have tested this on several devices. This is happening with 2.8.1, but works fine with 2.7. When the app freezes, I see memory spikes in my memory monitor. When the app unfreezes, everything works fine. I didn't notice a problem when testing with 20 or so regions. When testing with 300 regions, there is definitely an issue.

davidgyoung, 2016-06-12:
Thanks for this report, @marwfair.  I suspect the reason this is happening is that as of version 2.8, the library persists its monitored regions to a file to prevent spurious entry events across an app restart.  This probably is not performant for such a large number of regions.  

A few possible solutions:  (1) optimize the file persistence when starting to monitor a large number of regions.  (2) only persist a limited number of regions (3) allow disabling this persistence.    I want to add the third option for other reasons as well.

On a related note, aside from this issue, I am curious to hear how the library performs monitoring so many regions -- I have not tested it with this big of a number.

marwfair, 2016-06-13:
@davidgyoung The library seems to perform pretty well with lots of beacons. I just had a few issues with 2.7 not wanting to scan, especially after the app is resumed. That issue seemed to had been fixed in 2.8.1, but because of the app freezing, I had to use 2.7.

permutohedron, 2016-06-30:
Hi. The Monitoring status persistence can be disabled right now. Just call

`MonitoringStatus.getInstanceForApplication(applicationContext).stopStatusPreservationOnProcessDestruction();`

davidgyoung, 2016-07-22:
Fixed in #405

closed 2016-07-22 by davidgyoung

davidgyoung, 2016-07-25:
Released in 2.9