AltBeacon/android-beacon-library #384: Better support for custom beacons with extra data advertisement packets (BeaconParser extra layout)

mfatiga, 2016-06-13:
## Intro

I am developing an app for reacting to custom beacons. The beacon implementation broadcasts multiple different advertisement packets one of which is an identification packet (similar to iBeacon) and the other one is a telemetry packet (similar to Eddystone-TLM). According to [BeaconParser.setBeaconLayout](http://altbeacon.github.io/android-beacon-library/javadoc/org/altbeacon/beacon/BeaconParser.html#setBeaconLayout(java.lang.String)), this library supports parsing custom advertisement packets to extract extra data from them.

## Problem

This does not work in my project because of 2 reasons:

### 1 - no service UUID

`beacon.getServiceUuid()` returns -1 because of which my beacons are automatically dismissed by `GattBeaconTracker.track`, which then doesn't collect any additional telemetry packets for that beacon.

### 2 - advertisement packets don't contain the same byte sequence
The line in [BeaconParser.setBeaconLayout](http://altbeacon.github.io/android-beacon-library/javadoc/org/altbeacon/beacon/BeaconParser.html#setBeaconLayout(java.lang.String)) 
> Signifies that the layout is secondary to a primary layout with the same matching byte sequence (or ServiceUuid)

My beacon's advertisement packets don't contain same matching byte sequences so the library does not relate the 2 beacon parser layouts as being for the same beacon type.

## My current beacon layout example
```
BeaconParser main = new BeaconParser().setBeaconLayout("m:2-3=abcd,i:4-5,i:6-7,i:8-9,p:24-24");
BeaconParser extra = new BeaconParser().setBeaconLayout("x,m:2-2=51,d:21-21");
```
As you can see, the layout's don't have matching services or matching byte sequences.

## Solution proposals

The `beacon.getServiceUuid() != -1` condition in `GattBeaconTracker` could be removed and `GattBeaconTracker` renamed to `BeaconTracker` :-)

### Relate beacon parser beacon layouts by their identifiers
Beacon parser beacon layouts should not be related by matching byte sequences or service UUIDs, they should be related by the `BeaconParser.identifier` field. 

### Change the way extra layouts are handled
The extra data layout should be added separately to the `BeaconParser`, for example:
```
BeaconParser parser = new BeaconParser().setBeaconLayout("m:2-3=abcd,i:4-5,i:6-7,i:8-9,p:24-24");
parser.addExtraDataLayout("x,m:2-2=51,d:21-21");
```

## Compatibility
Removing the `beacon.getServiceUuid() != -1` condition in `GattBeaconTracker` should not pose any significant performance hits and should be backwards compatible.

The `GattBeaconTracker` beacon lookup table should work without issues because it uses the combo of service UUID and mac address.

Relating beacon layout's by their identifiers should not affect the current library functionality and should be backwards compatible.

Changing the way extra layouts are handled would break most Eddystone apps using this library but would be a prettier solution IMHO.

Please comment and indicate any flaws in my proposed solutions. Thank you!

davidgyoung, 2016-06-13:
Thank you for this well-considered proposal, @mfatiga.  These are good ideas.  My only hesitation is that I would not want to make a change that "would break most Eddystone apps".  

I like the idea of associating extra layouts programmatically, but the serviceId association must be retained for backward compatibility if it is to be put in before a 3.x release, which is still a ways off.  Perhaps the logic of a new `BeaconTracker` class could check for a programmatic association, and if one does not exist, but a serviceUuid does exist, fall back to the existing behavior where extra data are associated by serviceUUID.  This existing behavior would be considered deprecated, and perhaps there might be a configuration flag to turn it off (although I'm not certain there is a good reason to add such a configuration flag.)

mfatiga, 2016-06-13:
Thank you, @davidgyoung ! I agree with you, I will create a pull request with a proposed solution as soon as I get some free time.

mfatiga, 2016-06-17:
I have been testing the commit for the last couple of days, it's in production already and working great.

davidgyoung, 2016-06-17:
Good to hear.  Give me a few days to review the PR and to a little testing of my own.

davidgyoung, 2016-06-24:
resolved by #387

closed 2016-06-24 by davidgyoung

davidgyoung, 2016-07-25:
Released in 2.9