AltBeacon/android-beacon-library #385: Moto X 2nd gen distance coefficients

HappyDr0id, 2016-06-14:
Hello,

I have done measurements for the Moto X 2nd gen device, but, following your indications at http://altbeacon.github.io/android-beacon-library/distance-calculations2.html, I can't figure out how to determine the Intercept, Multiplier and Power values.

Here are my values :
Meters  RSSI
0.25 	-58
0.5 	        -69
1 		-77
2 		-87
3 		-92
4 		-96
5 		-98
6 		-96
7 		-91
8 		-98
9 		-95
10 		-96
12 		-101
14 		-99
16 		-100
18 		-98
20 		-97
25 		-98
30 		-101

Android device:
version: 6.0
build_number: MPE24.49-18
model: Moto X 2nd gen XT1092
manufacturer: Motorola

Beacon Info:
Kontakt.io Tag
Advertisements per second: 10
Transmit Power: Max 4dBm

IPhone 5 Average RSSI @1m: -60

Here is my spreadsheet calculations : https://drive.google.com/open?id=19N6ufipDyJMatF8v9iadVG5c6E-0fOEmTZ37wOKuPw0

Between Step 6 and Step 7 of the calculations tutorial, there is absolutely no clue given to determine this :
Distance formula coefficients calculated for these values:
 Intercept: ???
 Multiplier: ???
 Power: ???

Could you give some help please ? Thanks in advance.

HappyDr0id, 2016-06-15:
Nevermind, if I am not wrong, I understand that the Intercept value is C, the multiplier is A and the power is B.

davidgyoung, 2016-06-24:
Resolved by #386

closed 2016-06-24 by davidgyoung