AltBeacon/android-beacon-library #388: didExitRegion delay

medrockstar, 2016-06-16:
Hi,

I can't configure didExitRegion delay.

Tks,

davidgyoung, 2016-06-16:
I will take this as a feature request.  Let me know if you are interested in working on it.

See a possible workaround here:

http://stackoverflow.com/questions/30434823/is-it-possible-to-change-the-didexitregion-time-using-android-beacon-library

Rkr999, 2018-03-06:
Hi Team,
didExitRegion() not calling when exit from beacon.

davidgyoung, 2018-03-06:
@Rkr999, this is a different issue than described here.  This thread is about a feature request to make the exit region delay configurable.  If you are having trouble with the library, please open a new question on StackOverflow.com, as that is the appropriate forum for troubleshooting questions.