AltBeacon/android-beacon-library #389: Not able to stop Advertising

ashal2215, 2016-06-18:
###Expected behavior
	mBeaconTransmitter.stopAdvertising();
This has to stop advertising 

###Actual behavior
The mobile will keep on advertising after the stopAdvertising  method is called .

###Steps to reproduce this behavior
					mBeaconTransmitter.startAdvertising(beacon);
					Thread.sleep(10000);
					mBeaconTransmitter.stopAdvertising();
			

				
###Mobile device model and OS version
Moto G (3rd Generation)
OS : 5.1.1
###Android Beacon Library version
Android Beacon Library 2.7 
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2016-06-18:
Thanks for the report, @ashal2215.  Could you please turn on debug logging with beaconManager.setDebug(true); then capture LogCat output when advertising stopping fails?

ashal2215, 2016-06-22:
Thanks David , I could able to resolve the problem.

hassanAndro, 2016-12-20:
hey @ashal2215  can you please tell me how did you resolve this problem ? i am facing it too

davidgyoung, 2016-12-20:
@hassanAndro, I don't know how @ashal2215  resolved this.   Perhaps he can comment to let us know.

hassanAndro, 2016-12-20:
@davidgyoung Sir do you have any suggestions ?

davidgyoung, 2016-12-20:
@hassanAndro, can you please turn on debug logging with beaconManager.setDebug(true); then capture LogCat output when advertising stopping fails?

hassanAndro, 2016-12-20:
But this method is deprecated now

davidgyoung, 2016-12-20:
It still works despite the deprecation warning

hassanAndro, 2016-12-21:
I have scene something strange that if i advertise a android as beacon and call stop after 2 seconds in the locate app, i have tested that i will continuous to advertise for 10 to 11 secs ?

hassanAndro, 2016-12-21:
Can i decrease the interval of advertising in Altbeacon or any other method to advertise just for 5 seconds ? @davidgyoung

davidgyoung, 2016-12-22:
It is unusual for a device to continue advertising for 10 seconds after you call stop.    @hassanAndro, can you confirm that you also see this with the Locate app?   This may be device-specific behavior that you cannot control. 

 If you **do not** see this behavior with Locate, but you do see it with your own custom app, it is possible that it is an issue with how you are using the library.

hassanAndro, 2016-12-22:
@davidgyoung  i have tested and seen that advertisement with Locate App Sir and it remains on scanning screen for 10 to 12 seconds

davidgyoung, 2016-12-22:
Right, but if you use Locate to do the transmitting on the problematic Android device, does the Locate app also keep hitting the beacon for 10-12 seconds after hitting the off button?  

Keep in mind that when used as a detector, Locate keeps a beacon on the screen for ~10 secs after it is last detected.  You can tell when it actually stops being detected when the rssi stops changing.  You sure this is not what you are seeing?

supr84, 2017-03-09:
I am facing this problem, please help if someone has already solved this.

jeremycis, 2017-10-07:
Hi, I have the same issue with . It seems that something change the variable BeaconTransmitter.this.mStarted to false in BeaconTransmitter.class after enter in onStartSuccess and execute BeaconTransmitter.this.mStarted = true;

Can anyone help? If you need more data I can gave you.

Device: ASUS Zenfone ZE552KL
OS : 7.0
Android Beacon Library: 2.12.2

[LOGCAT_BEACON.txt](https://github.com/AltBeacon/android-beacon-library/files/1364207/LOGCAT_BEACON.txt)

davidgyoung, 2017-10-10:
@jeremycis, according to the log you attached, BeaconTransmitter is getting constructed over and over again with lines like this:

`10-06 22:39:53.151 14767-14767/com.cisquinidev.beacontx D/BeaconTransmitter: new BeaconTransmitter constructed.  mbluetoothLeAdvertiser is android.bluetooth.le.BluetoothLeAdvertiser@9ae8cea`

I suspect the instance of BeaconTransmitter that you are using to stop advertising is not the same one that you used to start advertising.  If I am correct, this is what would cause the problem.  You can't stop advertising on a different instance that used to start advertising.

jeremycis, 2017-10-10:
@davidgyoung, you are right! I was doing this inside my button.setOnClickListener callback. I put outside and now it works like a charm. Thank you!

jprashant610, 2018-03-07:
@davidgyoung i am facing the same issue of skipping stop advertising : not started

03-07 15:54:25.745 26859-26859/droidmentor.beacontransmitter D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
03-07 15:54:25.749 26859-26859/droidmentor.beacontransmitter D/BeaconTransmitter: new BeaconTransmitter constructed.  mbluetoothLeAdvertiser is android.bluetooth.le.BluetoothLeAdvertiser@c4ef15d
03-07 15:54:25.754 26859-26859/droidmentor.beacontransmitter D/BeaconParser: Identifier size is just right: 7f 80 ea ca 05 00 03 ac 52 41 c8 bd a6 93 2a 2f
03-07 15:54:25.755 26859-26859/droidmentor.beacontransmitter D/BeaconParser: Identifier size is just right: 00 01
03-07 15:54:25.755 26859-26859/droidmentor.beacontransmitter D/BeaconParser: Identifier size is just right: 00 01
03-07 15:54:25.757 26859-26859/droidmentor.beacontransmitter D/BeaconTransmitter: Starting advertising with ID1: 7f80eaca-0500-03ac-5241-c8bda6932a2f ID2: 1 ID3: 1 and data: 02 15 7F 80 EA CA 05 00 03 AC 52 41 C8 BD A6 93 2A 2F 00 01 00 01 C5  of size 23
03-07 15:54:25.763 26859-26859/droidmentor.beacontransmitter D/BluetoothAdapter: STATE_ON
03-07 15:54:25.774 26859-26872/droidmentor.beacontransmitter D/BluetoothLeAdvertiser: onClientRegistered() - status=0 clientIf=6
03-07 15:54:25.790 26859-26859/droidmentor.beacontransmitter D/BeaconTransmitter: Started advertisement with callback: org.altbeacon.beacon.BeaconTransmitter$1@35d10d7
03-07 15:54:25.790 26859-26859/droidmentor.beacontransmitter I/BeaconTransmitter: Advertisement start succeeded.
03-07 15:54:40.007 26859-26859/droidmentor.beacontransmitter D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
03-07 15:54:40.012 26859-26859/droidmentor.beacontransmitter D/BeaconTransmitter: new BeaconTransmitter constructed.  mbluetoothLeAdvertiser is android.bluetooth.le.BluetoothLeAdvertiser@c4ef15d
03-07 15:54:40.012 26859-26859/droidmentor.beacontransmitter D/BeaconTransmitter: Skipping stop advertising -- not started
03-07 15:54:40.014 26859-26859/droidmentor.beacontransmitter D/BeaconParser: Identifier size is just right: 00 80 ea ca 05 00 03 ac 52 41 c8 bd a6 93 2a 2f
03-07 15:54:40.015 26859-26859/droidmentor.beacontransmitter D/BeaconParser: Identifier size is just right: 00 01
03-07 15:54:40.015 26859-26859/droidmentor.beacontransmitter D/BeaconParser: Identifier size is just right: 00 01
03-07 15:54:40.017 26859-26859/droidmentor.beacontransmitter D/BeaconTransmitter: Starting advertising with ID1: 0080eaca-0500-03ac-5241-c8bda6932a2f ID2: 1 ID3: 1 and data: 02 15 00 80 EA CA 05 00 03 AC 52 41 C8 BD A6 93 2A 2F 00 01 00 01 C5  of size 23
03-07 15:54:40.018 26859-26859/droidmentor.beacontransmitter D/BluetoothAdapter: STATE_ON
03-07 15:54:40.023 26859-26871/droidmentor.beacontransmitter D/BluetoothLeAdvertiser: onClientRegistered() - status=0 clientIf=7
03-07 15:54:40.044 26859-26859/droidmentor.beacontransmitter D/BeaconTransmitter: Started advertisement with callback: org.altbeacon.beacon.BeaconTransmitter$1@70bddc4
03-07 15:54:40.045 26859-26859/droidmentor.beacontransmitter I/BeaconTransmitter: Advertisement start succeeded.

jprashant610, 2018-03-07:
thanks @davidgyoung , i have resolve the problem. Did same mistake like jeremycis .

davidgyoung, 2018-03-07:
Closing this issue, since all documented cases of this were caused by trying to stop advertising with a different instance of the transmitter than used to start it.

closed 2018-03-07 by davidgyoung