AltBeacon/android-beacon-library #390: when is didEnterRegion called?

parthrparekh93, 2016-06-20:
Hi, I am using the latest version 2.8.1 of Android Beacon Library. I wanted to know whether the function didEnterRegion is called only when a mobile device enters in the range of the beacon device or even after the device has entered the range once and still detected the same beacon once again?

Thanks

davidgyoung, 2016-06-20:
As of version 2.8 it is called only when a mobile device enters in the range of the beacon device for the first time.  It is not called again until the region is exited, even if the app is stopped and restarted.  This is similar to the way iOS behaves.

Prior to 2.8 it was called both each time the mobile device enters in the range of the beacon device after each app startup.  

Because of widespread confusion about this, a future release will make the library configurable to behave one way or the other.

closed 2016-06-20 by davidgyoung

parthrparekh93, 2016-06-20:
If it is not called again, then when will I get a notification? According to the code, if the didEnterRegion function is called again after boot and the application is in background, then we get a notification. Now, as per what you said, there is no way to get a notification.

davidgyoung, 2016-06-22:
You will get a second enter notification after the system detects that the beacon is no longer visible (e.g. no packets have been received in 30 seconds) and then it becomes visible again.