AltBeacon/android-beacon-library #393: Make monitoring state persistence configurable and fault-tolrant

davidgyoung, 2016-06-22:
**Background:**  As of library version 2.8, the state of region monitoring is persisted across app restarts.  If you are in a region and the app restarts, the library remembers that you are still in the region.   This is similar to the way iOS deals with beacons.  This change was implemented in #315. 

**Problems:**

While this change has benefited many users, it has caused some problems for other users:

1. During development, it is unexpected that when restarting the app, you don't get an entry callback.  This makes testing difficult.  There have been lots of support requests about this.
2. If you have an app that does not do background monitoring (it only monitors when the app is open), it may not be appropriate to re-load the state of the monitored regions, because the app may not have been running for a long time -- the app may not really know the state of each region despite what has been persisted
3. If there are a very large number of regions being monitored, the persistence takes a large amount of time and fails.  This problem was reported in #383 

**Proposed Solutions:**

1. Allow a configuration option to turn off monitoring persistence.  This would not be the default, but if set, it would make the library behave as before 2.8.
2. Record a timestamp when regions were persisted, and if there are no changes, touch this timestamp periodically when the app is running.  Before restoring region state on library startup, check this timestamp. If the timestamp is older than a certain time (say 15 minutes), then do not restore region state.
3. Limit the number of regions that are persisted.  If more than (say 20) regions are being monitored, don't persist any.  This is probably better than the alternative of persisting the first few regions, because that would lead to indeterminate behavior.

Any feedback on these proposed changes is appreciated, especially from @permutohedron who originally added the persistence and @marwfair who reported the problem with monitoring a large number of regions.

I wish I had a better idea for allowing new entry events during development.  Ideally, we could detect if a new copy of the app had been installed during development and then clear the persistence file.  But I don't know of any way to do this.  I only know how to detect if an app is run in development mode or if the formal version changes.   I don't know that it is possible to detect if the development version changes.

permutohedron, 2016-06-30:
Hello
Ad 1. As i mentioned in #383 it can be disabled right now, but I'll try to propose a better solution for that.
Ad 3.  I think the ability to turn off the persistence is sufficient. Everyone will be available to turn it off not only at the configuration time but even when the number of monitored beacons rises nondeterministically.
Ad 2. Right now status file is touched when something has changed (There is no other way to do it as there are absolutely no callbacks when Android kills the process) so periodical check would not change anything and could be even worse. What would be the the benefit of it?

davidgyoung, 2016-06-30:
Thanks for your thoughts, @permutohedron.  

For 1, Yes, it is good to hear this can already be disabled for folks who need to do so.   I would propose an additional wrapper method that is a bit friendlier.  Like `beaconManager.setRegioinStatePeristenceEnabled(false)`

For 2, The key part of the proposal is remembering how long it has been since the Android Beacon Library was last scanning.  If it has been > ~15 min, don't restore state.  The reason is that if you haven't been scanning in awhile, you really don't know the state of each region.  Somebody could have manually launched an app after it was not running for a long time, but the last time it was running it was in region.  In such cases, a new `didEnterRegion` event would be appropriate.

For 3, I think allowing an app freeze if somebody has not disabled persistence and has a large number of regions is not the best way to handle this.  If we know that persistence cannot reasonably handle saving a large number of regions, and we can see that they exist, why proceed with doing so?  On iOS, the system will only keep track of 20 regions, so it is not a completely foreign concept to limit the number.  An error can be put in the log telling folks that persistence has been disabled due to the large region count.

davidgyoung, 2016-07-22:
Fixed in #405

closed 2016-07-22 by davidgyoung

davidgyoung, 2016-07-25:
Released in 2.9