AltBeacon/android-beacon-library #395: Is there a method to get scanRecord ? or scan response?

SamuelY, 2016-06-26:
Hi 

As title,  how could i get our beacons scan response package ? 

or how could i get the scanRecord[] so that i can handle it by myself ?

A reply would be greatly appreciated.

davidgyoung, 2016-06-26:
The library allows passing on the scan response when a beacon is not matched.  You could force this to happen by removing all the beacon parsers, but then it might defeat the purpose of the library.  What do you want the library to do if you are handling the parsing youself?

See here for more info:

http://stackoverflow.com/questions/33937005/monitor-non-beacon-ble-devices-android-beacon-library

SamuelY, 2016-06-26:
I want to show  only the beacons that belongs to us, the identification information can be found in scan response package...

davidgyoung, 2016-06-26:
OK, so it sounds like you want to get ranging updates just like the library already gives you, but you want access to the scan response data that Android adds the the end of the scan data.  I think this is a reasonable feature request to access this data.  I'd be curious for any thoughts on how you would like to access it -- it seems to me the two possibilities are to (a) simply provide access to the raw scan data from the beacon record or (b) parse the PDUs (e.g. flags, manufacturer advertisement, scan response) and provide access to them as an array on the beacon object.

In the meantime, you can probably do this with the existing library by defining a custom beacon layout that adds an extra identifier at the end to pick up the raw scan response records as an extra identifier.  The example below shows how you can do this.  Note that in this example ID4 will return you the raw bytes of the scan response, including the PDU type and length at the start.  Also note that this solution will alter the behavior of the library so that when the scan response changes, you will temporarily see two instances of the same beacon in the ranging callback, because the differing fourth identifier will be treated as a distinct beacon.  Still, this may be good enough for you.

```    
@Test
    public void testAccessScanResponseAsId4() {
        LogManager.setLogger(Loggers.verboseLogger());
        org.robolectric.shadows.ShadowLog.stream = System.err;
        byte[] bytes = hexStringToByteArray("0201061bffaabbbeac2f234454cf6d4a0fadf2f4911ba9ffa600010004c5ff0102030405060708090a0b0c0d0e0f101112121415161718191a1b1c1d1e1f");
        BeaconParser parser = new BeaconParser();
        parser.setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25,i:26-56");
        Beacon beacon = parser.fromScanData(bytes, -55, null);
        assertEquals("mRssi should be as passed in", -55, beacon.getRssi());
        assertEquals("uuid should be parsed", "2f234454-cf6d-4a0f-adf2-f4911ba9ffa6", beacon.getIdentifier(0).toString());
        assertEquals("id2 should be parsed", "1", beacon.getIdentifier(1).toString());
        assertEquals("id3 should be parsed", "4", beacon.getIdentifier(2).toString());
        assertEquals("id4 should be parsed", "0x0102030405060708090a0b0c0d0e0f101112121415161718191a1b1c1d1e1f", beacon.getIdentifier(3).toString());
    }
```

SamuelY, 2016-06-27:
It's a good solution to customize layout of a beacon we want. It's enough for me .
Thank you very much.

SamuelY, 2016-06-27:
If we have defined a packet named manufacturing advertising that don't follow iBeacon Advertising Package format, how could i define bacon layout to get the package

for example, 
standard iBeacon package => AD type is 0xFF + ....

but our format of package
Length is 0x11 + AD type is 0x07 + AD data is Complete 128bit service UUID .......

davidgyoung, 2016-06-27:
@SamuelY, I don't think the `setNonBeaconLeScanCallback` works for your use case because it is just a pass-through.  It just forwards the raw scan callback for advertising packets that don't match any beacon filters to the callback.  For such packets, no library features are exercised.

For the format you are using the library code as-is will not match it, because it won't let you match AD type 0x07, it only matches the types below:

    public static final byte MANUFACTURER_DATA_PDU_TYPE = (byte) 0xff;
    public static final byte GATT_SERVICE_UUID_PDU_TYPE = (byte) 0x16;

In order to allow this, the `BeaconParser` would have to be modified to at least match type 0x07.  This is all possible, but not a trivial change.

Maartinsh, 2019-09-17:
Has this been updated/fixed/done?

davidgyoung, 2019-09-17:
This is an open feature request with no work started or planned.  Volunteers welcome.