AltBeacon/android-beacon-library #398: Possible to toggle scanning on ACTION_SCREEN_ON

Thomas101, 2016-07-05:
Would it be possible to listen in to `Intent.ACTION_SCREEN_ON` and run a scan as soon as the user turns the screen on? This would behaviour closer to replicating ios with its `CLBeaconRegion.notifyEntryStateOnDisplay`.

davidgyoung, 2016-07-05:
This is certainly possible and is a reasonable feature request.

 The intent filter would need to be dynamically added to the StartupBroadcastReceiver when the service starts by a different trigger.  Then, the CycledLeScanner would then need to be modified to kick off a scan cycle right when the screen goes on, as notified by the StartupBroadcadtReceiver.

If you would like to work on this, I would welcome it as a pull request.

yodutouf, 2016-07-20:
I am working on it. I think to do the pull request at the end of august