AltBeacon/android-beacon-library #402: Problems with Eddystone-URL transmittint and Chrome Physical Web

radzio, 2016-07-20:
I was trying to transmit Eddystone-URL using this code:

```
        try
        {
            byte[] urlBytes = UrlBeaconUrlCompressor.compress("http://google.com");
            Identifier encodedUrlIdentifier = Identifier.fromBytes(urlBytes, 0, urlBytes.length, false);
            ArrayList<Identifier> identifiers = new ArrayList<Identifier>();
            identifiers.add(encodedUrlIdentifier);
            Beacon beacon = new Beacon.Builder()
                    .setIdentifiers(identifiers)
                    .setManufacturer(0x0188)
                    .setTxPower(-7)
                    .build();
            BeaconParser beaconParser = new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_URL_LAYOUT);
            BeaconTransmitter beaconTransmitter = new BeaconTransmitter(getApplicationContext(), beaconParser);
            beaconTransmitter.setAdvertiseTxPowerLevel(AdvertiseSettings.ADVERTISE_TX_POWER_HIGH);
            beaconTransmitter.setAdvertiseMode(AdvertiseSettings.ADVERTISE_MODE_LOW_LATENCY);
            beaconTransmitter.startAdvertising(beacon, new AdvertiseCallback()
            {
                @Override
                public void onStartFailure(int errorCode)
                {
                    Log.e(TAG, "Advertisement start failed with code: " + errorCode);
                }

                @Override
                public void onStartSuccess(AdvertiseSettings settingsInEffect)
                {
                    Log.i(TAG, "Advertisement start succeeded.");
                }
            });
        }
        catch (MalformedURLException e)
        {
            Log.d(TAG, "That URL cannot be parsed");
        }
```

I am able to to see to transmitted beacon on my second device using this app: https://play.google.com/store/apps/details?id=com.uriio&hl=pl

But this beacon is not visible in Chrome Physical Web on my second device. When I've used BeaconToy app to transmit Eddystone-URL beacon my second device has found this beacon both in Chrome Physical Web and BeaconToy.

Do I have something in my transmitting code or is it a bug in altbeacon?

davidgyoung, 2016-07-20:
See #401.  A release candidate with this fix should be available this week.

radzio, 2016-07-21:
@davidgyoung  thanks for response!

davidgyoung, 2016-07-22:
Fixed via #401

closed 2016-07-22 by davidgyoung

davidgyoung, 2016-07-25:
Released in 2.9