AltBeacon/android-beacon-library #403: No Monitoring information after killing an application built with the minify mode

yodutouf, 2016-07-21:
The bug occurs on all minified version of the application

### Actual behavior
After killing a minified application, the application does not receive the Monitoring information (information about an entry and an exit from a region) 

### Steps to reproduce this behavior
1- Develop a simple application with a RegionBootstrap
2- install a minify version with the following gradle config
```
 buildTypes {
        debug {
            minifyEnabled true
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
```

3- Launch the application and kill it
4- The service restart
5- Activate a beacon 
-> the log shows that the beacon is detected but there is no information about an entry in a beacon region.

### Solution

The bug is due to Callback class.
After killing the application, in the case where the callback class is minified,
the intent is null when the **"call" method** is called.

There is two solutions:
* adding this line into your proguard configuration
` -keep class org.altbeacon.beacon.service.Callback { *; }`
* or updating the call method in the Callback class:

```java
public boolean call(Context context, String dataName, Parcelable data) {
        LogManager.d(TAG, "test if intent exits intent: %s - intentPackageName: %s", intent, intentPackageName);
        if(intent == null){
            initializeIntent();
        }
        if (intent != null) {
            LogManager.d(TAG, "attempting callback via intent: %s", intent.getComponent());
            intent.putExtra(dataName, data);
            context.startService(intent);
            return true;
        }
        return false;
    }
```
### Mobile device model and OS version
Nexus 5, Samsung S5

### Android Beacon Library version
2.8.1

davidgyoung, 2016-07-21:
Thanks for the great bug report, @yodutouf. 

Can you explain why solution 2 works?  Why is the intent null if the class is minified?

yodutouf, 2016-07-22:
The problem is with the deserialization process of the regionData. 

### Reason

When restoring the region from MonitoringStatus.restoreMonitoringStatus(), the method CallBack.readObject() is not called if the CallBack class is minified

### Complementary Observation

Looking more in detail to the code to understand the reason of the bug, there is a strange Callback initialization for the regionData. 
While in the BeaconService.startMonitoringBeaconsInRegion, there is a callback object as parameter, it's not transmitted to create the regionData.
In fact, the MonitoringStatus class is reacreating a callback...

#### Actual code

**> Beacon Service**
```java

  public void startMonitoringBeaconsInRegion(Region region, Callback callback) {
        LogManager.d(TAG, "startMonitoring called");
        monitoringStatus.addRegion(region);
        LogManager.d(TAG, "Currently monitoring %s regions.", monitoringStatus.regionsCount());
        mCycledScanner.start();
    }
```
**> Monitoring Status**
```java

public synchronized void addRegion(Region region) {
        if (mRegionsStatesMap.containsKey(region)) return;
        mRegionsStatesMap.put(region, new RegionMonitoringState(new Callback(mContext.getPackageName())));
        saveMonitoringStatusIfOn();
}
```

#### Code Suggestion 

**> Beacon Service**

```java
  public void startMonitoringBeaconsInRegion(Region region, Callback callback) {
        LogManager.d(TAG, "startMonitoring called");
        monitoringStatus.addRegion(region, callback);
        LogManager.d(TAG, "Currently monitoring %s regions.", monitoringStatus.regionsCount());
        mCycledScanner.start();
    }
```

**> Monitoring Status**
```java

public synchronized void addRegion(Region region, CallBack callback) {
        if (mRegionsStatesMap.containsKey(region)) return;
        mRegionsStatesMap.put(region, new RegionMonitoringState(callback));
        saveMonitoringStatusIfOn();
}
```

davidgyoung, 2016-08-06:
Good find, @yodutouf.  If you'd like to make a pull request for these changes, I'd gladly merge it.  I agree that the solution of "updating the call method in the Callback class" is best, and I also agree with your code suggestion for constructing the callback.

yodutouf, 2016-09-19:
Sorry I was in a long term holidays. Do you still need a pull request about these points ?

davidgyoung, 2016-09-19:
Please!

cupakromer, 2017-03-30:
Can this be closed now that #432 is merged?

davidgyoung, 2017-03-31:
Mixed per #432

closed 2017-03-31 by cupakromer