AltBeacon/android-beacon-library #410: iBeacon state outside at first didDetermineState invoke

hcknl, 2016-08-03:
Some iBeacons get monitored as outside at first run of my app, however they did not go to inside state or they got inside before app relaunched. Can this behaviour be a bug , seems like caching but it can't be caching ,i guess? and also sometimes monitored beacons found really really long time(maybe 1 min) after app launch and sometimes just 1 sec later.

I'm trying with iBeacon devices,iPhone 6 plus(AirBeacon app for transmitting),General Mobile 5 plus(Locate app for transmitting),Nexus 5x(my app for monitoring)

lib version 2.8.1

hcknl, 2016-08-04:
solved by stopping monitoring before remonitor again ! 

for (Region reg : beaconManager.getMonitoredRegions()) {
            println(reg.getUniqueId() + " " + reg.getId1().toString() + " cancelled");
            try {
                beaconManager.stopMonitoringBeaconsInRegion(reg);
            } catch (RemoteException e) {
                e.printStackTrace();
            }
        }

davidgyoung, 2016-08-05:
@hckni, as of library version 2.8, monitoring state is persisted across app restarts.  This helps prevent against multiple `didEnterRegion` events, but it can cause some confusion in cases like you describe.   Version 2.9 added some enhancements that makes this easier to work with.  You can read about these here: https://github.com/AltBeacon/android-beacon-library/pull/405.  This may help you so you don't have to stop monitoring before starting.

closed 2016-08-05 by davidgyoung

davidgyoung, 2016-08-06:
I added a page to the documentation to make this clearer: http://altbeacon.github.io/android-beacon-library/state-persistence.html

hcknl, 2016-08-09:
thank you for helping out with this confusion. I want to tell something new, with lib version 2.9 , monitored beacons got inside/outside very frequently, much than older versions.I just want to report that too.

davidgyoung, 2016-08-09:
@hcknl, any specific details on what you are seeing with unexpected entry/exit events in 2.9 would be appreciated as a new issue.

hcknl, 2016-08-10:
I've tested particularly on android 6.0.1(gm 5 plus) and android 7 developer preview 5(nexus 5x)
The problem is related to android 7 nougat.
Android 7 gatt service logs this "app is scanning too frequently"  and enter/exit events too much.
So it is not big deal , i guess. Forget about android N.

Thanks anyway.

davidgyoung, 2016-08-20:
@hcknl, I have opened an issue for the problem you reported on Android N here: #418, and have a PR to fix it in #419.  There is a binary pre-release build and instructions to try it in #419.  Please give it a shot and let me know if it resolves the issue for you.