AltBeacon/android-beacon-library #411: MonitoringStatus statePreservation.

eliaslecomte, 2016-08-04:
###Expected behavior

After a state preservation, beacons only trigger when the lastSeenTime > 10 s.

###Actual behavior

After a state preservation, beacons trigger again if firstSeenTime > 10 s.

###Steps to reproduce this behavior

Make a beacon trigger.
Wait 10 seconds.
Kill the app.
Let the beacon service restart itself.
Beacon will trigger again since lastSeenTime is updated in the model (RegionMonitoringState), **but not serialized to file**.

###Android Beacon Library version

Latest releease

eliaslecomte, 2016-08-04:
A possible fix could be, upon de-serialization, to set the lastSeenTime to System.currentTimeMillis() if inside is true.

davidgyoung, 2016-08-06:
Thanks for this report, @eliaslecomte.  I had not noticed this problem, but based on analysis I agree it must exist.  To state my understanding of the problem a different way:

After state restoration, for any Region with a state of "in", failure to immediately detect a beacon  in the first scan cycle will cause an exit event, rather than waiting 10 seconds to do so.

I agree with the proposed fix.  If you'd like to make a pull request for this, I'd be glad to merge it.  If you can do so, please also add a test for this in MonitoringStatusTest.java.   You can see how to run tests in the README.

eliaslecomte, 2016-09-03:
Hi, I want to create my PR, but I can't push my branch. Can I get permissions for this or am I doing something wrong?

davidgyoung, 2016-09-03:
@eliaslecompte, please see the "Pull Request Pro Tips" here: https://guides.github.com/activities/contributing-to-open-source/

The key is to "fork the repository and clone it locally." Your fork will have full rights to push changes.

Let me know if you have trouble.

eliaslecomte, 2016-09-04:
That was kinda logic :-), however it's the first time I contribute to an open source library. Created the PR for review. 👍

eliaslecomte, 2016-09-05:
Can be closed :D.

closed 2016-09-05 by davidgyoung