AltBeacon/android-beacon-library #412: Android 6.0.1 - security patch level 1 don't work

FedericoTorsello, 2016-08-07:
###Expected behavior
Scanning for detect the beacons nearby and getting information.

###Actual behavior
The scanning start but on **didRangeBeaconsInRegion** anybody beacon is detected.
The method _onRequestPermissionsResult_ is correctly set.
The Bluetooth work correctly in other applications (Bluetooth LE Scanner) that use classic _onLeScan(...)_ method.

Other test: **Locate** application don't detect nothing. 
Test with Locate:
1) bluetooth start disabled/enabled
2) wifi start disabled/enabled
no change detected...

###Steps to reproduce this behavior
Start application e start scanning, but the **size** of the variable beacons of "final Collection<Beacon> beacons" is _always zero_.

###Mobile device model and OS version
ASUS_Z00LD (http://specdevice.com/showspec.php?id=9275-2ed6-b72b-0cbf5429c89f), Android 6.0.1 - security patch level 1, API 23

###Android Beacon Library version
2.9

davidgyoung, 2016-08-07:
To clarify, it sounds like you are suggesting that this ASUS_Z00LD 6.0.1 OS image breaks scanning on the newer Android 5+ BLE scanning APIs.  Correct?

To confirm the library still works with the older 4.3 APIs, can you try adding the line:

  `beaconManager.setAndroidLScanningDisabled(true);`

And retesting to see if scanning works?

I will note that I have successfully scanned beacons with the library using the Android L APIs on the Nexus 9 and Nexus 5X with 6.0.1.

FedericoTorsello, 2016-08-08:
On the basis of my tests:

1. if in build.gradle(Module: app) I set: minSdkVersion 18 and targetSdkVersion _23_
  1. `beaconManager.setAndroidLScanningDisabled(true);` don't have effect. The scanning start but the Collection is always size zero.
  2. with the default config, the scanning start but the Collection is always size zero.

2. if in build.gradle(Module: app) I set: minSdkVersion 18 and targetSdkVersion **18** the app works without any problems.

**It would seem a problem connected to the permissions of Android 6.**

The code I used is: [copy and paste of _android-beacon-library-reference_ project]
```
	@Override
	public void onRequestPermissionsResult(int requestCode,
										   String permissions[], int[] grantResults) {
		switch (requestCode) {
			case PERMISSION_REQUEST_COARSE_LOCATION: {
				if (grantResults[0] == PackageManager.PERMISSION_GRANTED) {
					Log.d(TAG, "coarse location permission granted");
				} else {
					final AlertDialog.Builder builder = new AlertDialog.Builder(this);
					builder.setTitle("Functionality limited");
					builder.setMessage("Since location access has not been granted, this app will not be able to discover beacons when in the background.");
					builder.setPositiveButton(android.R.string.ok, null);
					builder.setOnDismissListener(new DialogInterface.OnDismissListener() {

						@Override
						public void onDismiss(DialogInterface dialog) {
						}

					});
					builder.show();
				}
				return;
			}
		}
	}
```
P.S.
**Locate Beacon** (https://play.google.com/store/apps/details?id=com.radiusnetworks.locate&hl=en) does not work exactly such as my application.

davidgyoung, 2016-08-08:
Do you get prompted for location permission?  What do you see if you go to Settings -> Locate -> Permissions?  (Same question for your app.). Is location permission on?

FedericoTorsello, 2016-08-08:
Ok, **my app work** if I set `On` in `Settings -> Location` (_High accuracy_ or _Battery saving_ mode).

It is not the same for AltBeacon Loc. I have reinstalled him, but something goes wrong.
I checked in `Settings -> App -> AltBeacon Loc -> Permissions` and Location is set on.

I'm happy to have solved, but I really feel AltBeacon Loc does not work.

Thank You!

closed 2016-08-15 by FedericoTorsello