AltBeacon/android-beacon-library #413: It's not possible to schedule CycledLeScanner less than 5 minutes

brunocarneiro, 2016-08-09:
###Expected behavior
I would like to configure my app to scan for beacons every minute (or between 2 minutes).

###Actual behavior
The problem I've seen is that in the class CycledLeScanner five minutes is hard-coded. Look at this code: https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/service/scanner/CycledLeScanner.java#L294

davidgyoung, 2016-08-09:
Hi, @brunocarneiro.

You are correct that there is a hard-coded alarm time of 5 minutes, but this isn't what starts each scan cycle.  It is what re-starts scanning if the operating system (or the user) terminates the app.  A five minute threshold seems reasonable because this gives the OS time to finish memory intensive operations in case the app has been stopped due to low memory.

I think it is a fair feature request to make that five minute period configurable, however I would advise against lowering it significantly because of the consequences on quickly restarting the app when it is in a low-memory state.

If what you really wan to do is set the scan period, there are configurable settings for this:

```
beaconManager.setBackgroundScanPeriod(...)
beaconManager.setBackgroundBetweenScanPeriod(...)
beaconManager.setForegroundScanPeriod(...)
beaconManager.setForegroundBetweenScanPeriod(...)
```
Understand that the five minute period that is hard-coded above only gets triggered if the scanning process shuts down.

brunocarneiro, 2016-08-09:
Humm good to know @davidgyoung !
In fact I'm testing beacon monitoring in my app. So in some of my test cases I need to terminate the app.

Also, the feature I want to implement is notify the user if he's near a known beacon, even though the app is closed or running in background.
So, I guess this configuration would be awesome to my case.

Thanks for your reply! I really appreaciate!

davidgyoung, 2016-08-09:
Understood.  If this is an enhancement you need soon and would like to put together a pull request, I would gladly review and merge when ready.