AltBeacon/android-beacon-library #416: didDetermineStateForRegion() ambiguity

leeweijie, 2016-08-17:
When I call requestStateForRegion(), I cannot differentiate if the callback in didDetermineStateForRegion() is due to the request or due to the initialization of monitoring. Maybe add a boolean parameter to didDetermineStateForRegion()?

davidgyoung, 2016-08-17:
These APIs are designed to be similar to those available on iOS and other beacon detection platforms, and those don't offer such a flag. 

For the rare cases where you do need to know if the callback was based on a requestStateForRegion() call, it seems a reasonable alternative without changing the API would be to set a flag in  your app code and then check that flag in the callback.  If the flag is set, you know that is what caused the call and you can clear it.

leeweijie, 2016-08-17:
Ok thank you

closed 2016-08-17 by leeweijie