AltBeacon/android-beacon-library #417: Automatic background mode for bootstrapnotifier

leeweijie, 2016-08-17:
Currently it is very messy for me to manually turn off background mode when app is active. Many things to check (e.g. beaconmanager not assigned before RegionBootstrap is instantiated). Maybe add a bool to contructor of RegionBootstrap for auto background mode?

davidgyoung, 2016-08-17:
The `RegionBootstrap` does not control automatic background mode switching.  That is handled by the `BackgroundPowerSaver` class.  If want automatic background mode switching, you construct that class, if you don't want it, you don't.

Are you simply proposing that instead of doing this a boolean flag be added to the `RegionBootstrap` that does this internally?

leeweijie, 2016-08-17:
Thank you, didnt know that BackgroundPowerSave does that

closed 2016-08-17 by leeweijie