AltBeacon/android-beacon-library #418: Android N Repeat Region Exit/Entry Events

davidgyoung, 2016-08-20:
As of Android N, DP4:

> We’ve changed the BLE Scanning behavior starting in DP4. We’ll prevent applications from starting and stopping scans more than 5 times in 30 seconds. For long running scans, we’ll convert them into opportunistic scans.

https://developer.android.com/preview/support.html

With default settings, scanning is stopped and restarted every 1.1 seconds in the foreground (needed for older Bluetooth chipsets and stacks, mostly on devices running older Android version, that only deliver one advertisement callback per device per scan cycle.  Because mostly older devices behave this way, this regular scan restarting probably not needed for devices running Android N.)  

The problem happens on Android N because after 5.5 seconds, (5 scan restarts within 30 seconds) scans get blocked for the next 24.5 seconds, causing a region exit because no beacon packets are detected in that time.  After 30 seconds, a scan succeeds in starting and a new packet detection causes a new entry event.

Proposed solution is to detect if running on Android N, and if so, don't stop scans so frequently, and instead simply continue scanning.

hcknl, 2016-08-22:
Hello

Seems no trouble at all, works very well on Android N right now  but i wonder that, may continuous scanning cause bluetooth device problem or battery efficiency problem?

davidgyoung, 2016-08-22:
Thanks for the test.  Continuous scanning does use more battery, which is why it is recommended (and by default) only done by the library when the app is in the foreground.

glose25, 2016-08-22:
This fixes the issue I was seeing on Nexus 6P with N Preview 5. Thanks.

SteveReeves, 2016-10-04:
We're having this issue on our 7.0 devices running AltBeacon and I just wanted to clarify what the solution here is.  Do I simply need to change the foreground scan period to something longer (like 6.1 seconds) on Android N devices?

eliaslecomte, 2016-10-04:
AltBeacon too often stops - starts a BLE scan. By using the latest AltBean, this stop-starting does not happen as frequently on Android N. So using 2.9.1 or later should help you.

SteveReeves, 2016-10-04:
I updated to 2.9.1 yesterday and I'm still seeing what David mentioned above where I get results returned for the first 5.5 seconds, but the subsequent 24.5 seconds has no beacons showing up in scans.

davidgyoung, 2016-10-04:
@SteveReeves, two questions:

1. Have you customized your scan rate with the library?
2. Can you please turn on debug logging,  reproduce this problem, then send me a LogCat excerpt?  `beaconManager.setDebug(true)`

eliaslecomte, 2016-10-04:
Also, Android throws errors in Logcat if it is blocking scans.

SteveReeves, 2016-10-04:
@davidgyoung 
1. I customized the background scan period and the background between scan period.  The foreground scan periods I left untouched.
2. I restricted the LogCat output to only my app and let it run for a few minutes.
[AltBeaconLogs.txt](https://github.com/AltBeacon/android-beacon-library/files/509188/AltBeaconLogs.txt)

@eliaslecomte - I am seeing log lines such as this: 
10-04 13:53:52.895 6661-7412/? E/BtGatt.GattService: App 'com.net.app' is scanning too frequently 

As an aside - I realized this morning that version 2.9.1 of AltBeacon has deprecated the setMonitorNotifier and setRangeNotifier methods but I can't see what would have replaced them, and the code samples on your website still refer to these 2 methods.  Not sure if this is related or not.

Thanks.

davidgyoung, 2016-10-04:
@SteveReeves,  the code is behaving as if it doesn't think you are on Android N, or your are not on library version 2.9.1.  

Can you please add this to your code and let me know what it prints out in LogCat?

    Log.d(TAG, "This is Android Beacon Library version: "+org.altbeacon.beacon.BuildConfig.VERSION_NAME)
    Log.d(TAG, "This is Android version N? :"+(android.os.Build.VERSION.SDK_INT >= Build.VERSION_CODES.N));

SteveReeves, 2016-10-05:
@davidgyoung 
The output is as follows:
10-05 09:40:41.513 27594-27594/? D/BeaconMonitoringService: This is Android Beacon Library version: 2.9.1
10-05 09:40:41.513 27594-27594/? D/BeaconMonitoringService: This is Android version N? :true

Were there any changes I needed to make upon upgrading to 2.9.1? I see setMonitorNotifier and setRangeNotifier are both deprecated now?

davidgyoung, 2016-10-05:
You don't need to make any changes for 2.9.1.  `setMonitorNotifier` has been deprecated in favor of `addMonitorNotifier`, because you can have more than one.  But it still works the same way, and the deprecated versions still work fine, too.

For some reason, in your case the code shown below is falling into the else block when I would expect it to execute the if block.  As a result, it is stopping scanning when it should just leave it going. 

                    long now = System.currentTimeMillis();
                    if (android.os.Build.VERSION.SDK_INT >= Build.VERSION_CODES.N &&
                            mBetweenScanPeriod+mScanPeriod < ANDROID_N_MIN_SCAN_CYCLE_MILLIS &&
                            now-mLastScanStopTime < ANDROID_N_MIN_SCAN_CYCLE_MILLIS) {
                        // As of Android N, only 5 scans may be started in a 30 second period (6
                        // seconds per cycle)  otherwise they are blocked.  So we check here to see
                        // if the scan period is 6 seconds or less, and if we last stopped scanning
                        // fewer than 6 seconds ag and if so, we simply do not stop scanning
                        LogManager.d(TAG, "Not stopping scan because this is Android N and we" +
                                " keep scanning for a minimum of 6 seconds at a time. "+
                                 "We will stop in "+(ANDROID_N_MIN_SCAN_CYCLE_MILLIS-(now-mLastScanStopTime))+" millisconds.");
                    }
                    else {
                        try {
                            LogManager.d(TAG, "stopping bluetooth le scan");
                            finishScan();
                            mLastScanStopTime = now;
                        } catch (Exception e) {
                            LogManager.w(e, TAG, "Internal Android exception scanning for beacons");
                        }
                    }

Based on your setup, I believe ` mBetweenScanPeriod+mScanPeriod` is 1100, which should cause the first part of the if statement to evaluate to true.  (ANDROID_N_MIN_SCAN_CYCLE_MILLIS is 6000 milliseconds.)  The second part should also evaluate to true, because according to the log you sent, the scans are being stopped every 1100 ms.  

Unless you see something I don't, I may have to provide a special build that adds extra debug logging that breaks out the terms to see why this if statement is not evaluating to true.

SteveReeves, 2016-10-05:
@davidgyoung - I can't think of anything else I would be doing that would cause this issue.  The only scan periods I'm changing are as follow:

beaconManager.setBackgroundScanPeriod(5000);
beaconManager.setBackgroundBetweenScanPeriod(30000);

Besides that I turn background mode on and off when required.

If you're able to provide a custom build with extra debugging info, I'd be happy to give it a shot.

Thanks again,
Steve

davidgyoung, 2016-10-05:
But for the log excerpt you provided, is the app in the foreground?  I see
scans going off every 1100 ms.

SteveReeves, 2016-10-05:
Yes, I leave the foreground scan periods as is.  The log excerpt I posted was when the app is running in the foreground.  I manually detect when the app moves to the background and turn on background mode which then uses the scan periods above.

davidgyoung, 2016-10-06:
@SteveReeves, I have built  a debug version of the library [here](https://github.com/AltBeacon/android-beacon-library/releases/tag/android-n-debug) that will output extra debug info to show what is going on.  Please try this on your device and look for output like below, which is what is shown running the reference application on a Nexus 9 with Android 7.0:

```
10-06 09:52:07.777 23080-23080/org.altbeacon.beaconreference I/BeaconService: beaconService version 2.9.1-11-gb3ac622-SNAPSHOT is starting up
...
10-06 09:45:46.090 17914-17914/org.altbeacon.beaconreference D/CycledLeScanner: --- Android N Scan Debugging ---
10-06 09:45:46.090 17914-17914/org.altbeacon.beaconreference D/CycledLeScanner: Build Version comparison: 24 >= 24 -> true
10-06 09:45:46.090 17914-17914/org.altbeacon.beaconreference D/CycledLeScanner: Scan Period Comparison: 0 + 1100 < 6000 -> true
10-06 09:45:46.090 17914-17914/org.altbeacon.beaconreference D/CycledLeScanner: Last scan comparison: 1475761546090 - 1475761541642 < 6000 -> true
10-06 09:45:46.090 17914-17914/org.altbeacon.beaconreference D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 1552 millisconds.
10-06 09:45:46.090 17914-17914/org.altbeacon.beaconreference D/CycledLeScanner: starting a new scan cycle
```

After trying this, please post a comment on this thread 

with a log excerpt like the above.

SteveReeves, 2016-10-06:
@davidgyoung - I included the log line that checks the library version number because I noticed it says it's starting version 2.7 of the BeaconService.  It definitely seems to be using your custom debug build though...

```
10-06 11:42:28.527 17162-17162/com.net.app D/BeaconMonitoringService: This is Android Beacon Library version: 2.9.1-11-gb3ac622

10-06 11:42:28.545 17162-17162/com.net.app I/BeaconService: beaconService version 2.7 is starting up
```

I don't see any of your other log lines referenced above...

davidgyoung, 2016-10-06:
@SteveReeves, I think this is an important clue.  I suspect you have to different copies of the Android Beacon Library bundled with your app.   This would explain both the what you see above and the mysterious fact that you are using 2.9.1 but you don't see the Android N-specific scanning behavior that exists in that version.

Are you using any other libraries that may have bundled with the Android Beacon Library 2.7 behind the scenes?

To find, this you might try running the following on your command line to search your build folder to see what files may include it.  In my example below, you can see that I matched android-beacon-library_065301b53fcfbfce5e7398f3f1723953247bd8a7.  You might see if you match other libraries, too.

```
$ egrep -r "beaconService version" app/build
Binary file app/build/intermediates/instant-run-support/debug/classes.zip matches
Binary file app/build/intermediates/transforms/dex/debug/folders/1000/10/android-beacon-library_065301b53fcfbfce5e7398f3f1723953247bd8a7/classes.dex matches
Binary file app/build/intermediates/transforms/dex/release/folders/1000/1f/main/classes.dex matches
```

SteveReeves, 2016-10-06:
@davidgyoung I figured that would be a significant clue as to what's going on.  Unfortunately when I run your command, I only see 1 match:
```
Binary file my-sdk/build/tmp/expandedArchives/classes.jar_9h2z6qbouutiao98wg1d37cdf/org/altbeacon/beacon/service/BeaconService.class matches
```

Perhaps I should mention the AltBeacon library is being used in an SDK I'm writing, not directly in an app.  The app itself is a very simple test app that exercises my SDK.

davidgyoung, 2016-10-06:
when you run the `egrep` command, are you in your SDK project or app project?  I would certainly do it in the app project.

Another thing you can try is to unzip the APK generated for your app (yes, an APK is just a zipfile, although you may have to change the extension to let an unzip utility open it).   The you can run the same `egrep` command against the APK and see what it finds.

SteveReeves, 2016-10-06:
@davidgyoung - well, this is embarrassing. It's my own issue, and to do with my POM file for local installs of the SDK for testing.  It's now correctly using 2.9.1.

Sorry for the wild goose chase, but I really appreciate the help.

After running it for a few minutes, it's definitely working better, but I noticed something a little strange.  Around the 29s mark it stops seeing beacons for roughly the next 6.5 seconds.  No beacon region exit events are triggered during this time, but it will callback with 0 beacons found.  I've never seen this on any other phone.  It seems to be reproducible after moving from background mode to foreground mode, or during a first run after a fresh install.

davidgyoung, 2016-10-06:
Glad to hear you found the issue, @SteveReeves.  For the 6.5 second dropout you mention, this might be caused by a flaw in the algorithm during a background/foreground transition.  If you can capture this in a log, I'd appreciate it.  However, I really need it to include operating system log events as well because these give you the OS warnings about performing too many scans.

SteveReeves, 2016-10-07:
@davidgyoung - I figured that might be the case.  I've captured the logs (including system logs) - 
[AltBeacon_AndroidNLog.txt](https://github.com/AltBeacon/android-beacon-library/files/516670/AltBeacon_AndroidNLog.txt).  It sees the beacon at 11:33:24.212 but then not again until 11:33:30.944.  Let me know if there's anything else I can provide or help with. 

Thanks again

omanthey, 2016-11-07:
Hello,

@davidgyoung - In your introducing post you speak of this problem as the "default behaviour" in Androin N. Is there any way to "tweak" this behaviour?

I developed an app relying on Peter Metzs IBeacon plugin where I determine the location of an device within a museums exhibition using beacons.

While working quit good with Lollipop & Marshmallow devices the accuracy using a Nougat device is really bad as the exhibition visitor passes a couple of beacons (where events should be triggered) without notifing ...

Are there any workarounds to solve this kind of problem?

Thanks,
- Oliver Manthey

AlexLardschneider, 2016-11-13:
I have the same problem, scanning every 3 seconds. One iBeacon is in reach (phone running 6.0.1 finds it without problems), but the phone running 7.1 keeps having problems scanning. I updated to 2.9.1, enabled debug mode and captured some LogCat output in the file below.

The log is getting spammed by `CycledLeScannerForLollipop: Scan Failed`, and events like `didEnterRegion`, `didExitRegion` or `didRangeBeaconsInRegion` are not getting called. The code works fine below Android N.

I attached a sample log below, including system log calls like `bt_att`.
[Logcat.txt](https://github.com/AltBeacon/android-beacon-library/files/587746/Logcat.txt)

davidgyoung, 2016-11-14:
@AlexLardschneider, reviewing your log, it appears that it is showing a different problem than described by this thread.  In your case, the repeated error is not about scans being blocked by Android N:

    11-13 13:26:26.884 23501-23589/com.android.bluetooth E/bt_att: GATT_Register: can't Register GATT client, MAX client reached!
    11-13 13:26:26.884 23501-23589/com.android.bluetooth E/bt_btif: Register with GATT stack failed.

There is a bug report for this described here:  https://code.google.com/p/android/issues/detail?id=221402  If you want to discuss this issue further, please open a different issue.

AlexLardschneider, 2016-11-14:
@davidgyoung Ok, thanks for letting me know.

hirenpatelazilen, 2017-05-20:
@davidgyoung , first appreciate your effort for this great library.

I found an issue, beacons are getting enter/exit frequently within 10 seconds on some devices(Samsung 5.0 and Nexus 5X 6.0.1)

What I did:

```
iBeaconManager = BeaconManager.getInstanceForApplication(this);
iBeaconManager.bind(this);
iBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"));
iBeaconManager.setBackgroundBetweenScanPeriod(5000l);
iBeaconManager.setBackgroundScanPeriod(1000l);
```

Can you please let me know if any solution for the same.

Regards,
Hiren Patel

davidgyoung, 2017-05-22:
@hirenpatelazilen Please see #512

davidgyoung, 2017-06-30:
Fixed in #419

closed 2017-06-30 by davidgyoung