AltBeacon/android-beacon-library #420: SecurityException: Too many alarms

SteveReeves, 2016-08-25:
We're seeing a few (roughly 25 at this moment) Samsung devices running into an issue with the alarms manager.  I see that you put in a fix for this issue back in 2.3.  We're running 2.7 and seeing what looks to be the same issue. 

Here's the stacktrace:
`Fatal Exception: java.lang.SecurityException: !@Too many alarms (500) registered from pid 2480 uid 10080
       at android.os.Parcel.readException(Parcel.java:1540)
       at android.os.Parcel.readException(Parcel.java:1493)
       at android.app.IAlarmManager$Stub$Proxy.set(IAlarmManager.java:206)
       at android.app.AlarmManager.setImpl(AlarmManager.java:428)
       at android.app.AlarmManager.set(AlarmManager.java:215)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.setWakeUpAlarm(CycledLeScanner.java:299)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.setScanPeriods(CycledLeScanner.java:100)
       at org.altbeacon.beacon.service.BeaconService.setScanPeriods(BeaconService.java:298)
       at org.altbeacon.beacon.service.BeaconService$IncomingHandler.handleMessage(BeaconService.java:158)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:145)
       at android.app.ActivityThread.main(ActivityThread.java:5835)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)`

All the devices we're seeing this issue on are Samsung devices, running 5.0, 5.0.1, 5.1.1 or 6.0.1 and all of them are while the app is not in focus (ie: it's in the background).

Any ideas?  Unfortunately I'm not able to get my hands on any of the devices this is currently happening on.  I'm attempting to get more information from our partner regarding which models of Samsung phones this affects, but I know we're seeing it on Note 3/4 variants for sure.

Thanks, 
Steve

davidgyoung, 2016-08-25:
@SteveReeves, I'm surprised to hear this report.

Details on the change in 2.3  are here:  https://github.com/AltBeacon/android-beacon-library/issues/184

I personally tested this on a Samsung device with 5.0.2 and verified that this exception did not happen in the case I tested, which was waiting to reschedule over 500 alarms.    The solution was to only set one alarm, and reschedule it over and over again. 

 Reviewing the code in 2.7, this code is still there.  I can confirm that it does look like you are running 2.7, as in that version the alarm is reset on line 299 of CycledLeScanner.java.  

Two possibilities:

1.  Is your app starting and stopping the BeaconService regularly?  It looks like the way the code is written, if it stops and restarts scanning 500 times, this may happen.

2. Does another component of your app register a large number of alarms?  If so, whatever component most registers Alarm #500 will cause this problem in Samsung devices.

I suspect this may be an app-specific implementation issue because this is the first report I heard since the fix was put in last year, and before that fix I heard lots of reports of this problem.  The fact that you have seen it on several OS versions also suggests that it isn't just one strange device.  

One other thing that might help you.  There is a Samsung device lab you can use to remotely test your app on Samsung devices for free.  http://developer.samsung.com/remotetestlab/rtlDeviceList.action  You don't need beacons in the vicinity -- you can just run your app and wait for a crash.  Clearly, though, if it takes a long time to happen this will make using that solution difficult.

SteveReeves, 2016-08-26:
Thanks for your response.

I believe in certain circumstances we might be creating and destroying the BeaconManager instance we have fairly often while ranging beacons.  It sounds as though this could be causing the issue we see.  I'll take a look at ways we can mitigate it and give it a test in the Samsung device lab.  

Thanks again,
Steve.

davidgyoung, 2016-08-26:
Sounds like a plan.  I will mark this as a bug, because even if you are creating and destroying the scanning service periodically, this should not happen.  It would be possible to prevent.  But if another component is setting 499 alarms, this is not going to solve the problem.

SteveReeves, 2016-08-29:
Hi David, 

I just wanted to quickly clarify what you meant. 

> Is your app starting and stopping the BeaconService regularly? It looks like the way the code is written, if it stops and restarts scanning 500 times, this may happen.

When you say starting and stopping the BeaconService, do you mean ranging, monitoring or does it not matter? Or do you simply mean creating and destroying a BeaconManager instance?  

I'm trying to determine the best way to attempt to reproduce this so that I can hopefully take some steps to mitigate it from happening, and to also understand ways that I can avoid it in the future.

Thanks, 
Steve

davidgyoung, 2016-08-29:
The scanning service gets **started** whenever your app does both of the following:

1. Calls  beaconManager.bind(...)
2. Calls beaconManger.startRangingBeaconsInRegion(...) -OR- beaconManger.startRangingBeaconsInRegion(...)

It will get **stopped** whenever your app stops monitoring and ranging all regions (e.g. calls beaconManager.stopMonitoringBeaconsinRegion(...) -OR- beaconManager.stopRangingBeaconsInRegion(...) for all regions)

If this happens 500 times after the app launches, I believe it could cause the alarm exception you describe.

If, however, you are using the `RegionBoostrap` class, then this problem should not happen -- the beacon service should start up with your app and continue running as long as your app is running, preventing the alarm exception from happening at all.

davidgyoung, 2019-05-29:
cannot reproduce.  closing due to inactivity

closed 2019-05-29 by davidgyoung